ros2/rviz #578: LaserScan showing [0] points from [0] messages

HemaZ, 2020-07-19:
Hi, 
I'm having a problem visualizing LaserScan in Rviz2
I have fixed the frame as the LaserScan frame "link_1"  but in Status it shows "showing [0] points from [0] messages "
also, I've made sure that the topic currently has messages being published as you can see from the right terminal. 

[image]

I'm using the following gazebo ros plugin in my gazebo model.

```
 <sensor name='sensor_ray' type='ray'>
        <pose frame=''>0.5 0 0 0 -0 0</pose>
        <ray>
          <scan>
            <horizontal>
              <samples>200</samples>
              <resolution>1</resolution>
              <min_angle>-1</min_angle>
              <max_angle>1</max_angle>
            </horizontal>
            <vertical>
              <samples>1</samples>
              <min_angle>0</min_angle>
              <max_angle>0</max_angle>
            </vertical>
          </scan>
          <range>
            <min>0.05</min>
            <max>10</max>
          </range>
        </ray>
        <always_on>1</always_on>
        <visualize>1</visualize>
        <update_rate>100</update_rate>
        <plugin name='laser' filename='libgazebo_ros_ray_sensor.so'>
          <ros>
            <namespace>/simple_mr</namespace>
            <remapping>~/out:=laser_scan</remapping>
          </ros>
          <output_type>sensor_msgs/LaserScan</output_type>
        </plugin>
      </sensor>
```

Tested on:
-   Ubuntu 18.04.4
-   ROS2 eloquent

Rviz2 logs:

> [INFO] [rviz2]: Stereo is NOT SUPPORTED
> [INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
> [INFO] [rviz2]: Stereo is NOT SUPPORTED
> [INFO] [rviz2]: Stereo is NOT SUPPORTED

This is RViz version Alpha (ROS 2.0).
Compiled against Qt version 5.9.5.
Compiled against OGRE version 1.12.1 (Rhagorthua)

olmerg, 2020-07-31:
I have the same problem with this and other sensors, which I understand is that the problem is quality of service policy of sensor in gazebo plugins.

I think that the problem is the qos in the plugins because the plugins that works does not use rclcpp::SensorDataQoS , but my knowlegde abut ros2 is beginner level(nothing about QoS).

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_p3d.cpp#L123
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_ray_sensor.cpp#L110

and the plugins that works:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_p3d.cpp#L123
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_joint_state_publisher.cpp#L142

I assume this by this thread https://answers.ros.org/question/332207/cant-receive-data-in-python-node/

I am using ros Foxy updated from apt-get. I report the (issue  in dolly example)[https://github.com/chapulina/dolly/issues/16#issuecomment-667178225].

I have tried to change to the policies required for [sensorQos ](https://github.com/ros2/rmw/blob/9f2e181306c66ac0f09674086a108d4657611766/rmw/include/rmw/qos_profiles.h#L51-L62) in rviz (5, keep_last, Best Effort and Volatile) but this does not solve the problem.

ivanpauno, 2020-08-27:
As commented by @olmerg the problem is the qos settings.
The ray plugin publisher is using [lossy qos](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/2e4ddadef6aeb1dc0ebe5de3500270c07bd7ca19/gazebo_plugins/src/gazebo_ros_ray_sensor.cpp#L110), and the rviz subscription is requiring reliable communication.

@HemaZ did you try using "best effort" reliability?

ivanpauno, 2020-08-27:
Based on the comment [here](https://github.com/chapulina/dolly/issues/16#issuecomment-674422515), it seems that the rviz configuration needs to be stored in a file and rviz reopened.

HemaZ, 2020-08-27:
> As commented by @olmerg the problem is the qos settings.
> The ray plugin publisher is using [lossy [...]

tried "best effort" and "system default" and both works now. thank you

ivanpauno, 2020-08-27:
> system default

Bear in mind that system default is "unspecified", and depending on the rmw implementation (and other stuff ...), it might be reliable or best effort.
So prefer picking "best effort".

closed 2020-08-27 by ivanpauno