ros2/rviz #585: Map visualization fails if fixed frame is not picked first

pfedom, 2020-07-23:
I am not sure if this is a bug or if I am misunderstanding something. 
When I start slam_toolbox and want to visualize the created map in Rviz2 I need to first pick the fixed frame map and than add the map by topic. 
If I first add the map and than change the fixed frame the map doesnt get visualized. 

Only the order differs in both scenarios everything else is the same.

wjwwood, 2020-08-07:
This is likely a bug in the map display where when changing the reference frame it doesn't display the previous state in the new frame and instead waits for the map to be sent again. I imagine that enabling/disabling the display after changing frames will also not work but could be tested.

Offhand I don't know for sure what the cause is, but I'll place this as "help wanted".

pfedom, 2020-08-07:
completely forgot about this issue. 
I had a time missing in my odom transform message and now its working fine. 
I think its related to the bug I described but I am not sure. 

I think this issue can be closed, because it was an error in my setting.

closed 2020-08-07 by pfedom