ros2/rviz #602: Rviz2 crashes at launch time (symbol lookup error, undefined symbol)

anaelle-sw, 2020-08-25:
## Bug report

| Required Info | |
|---|---|
|Operating System|Ubuntu 20.04|
|Installation type|Binaries|
|Version or commit hash|Foxy|

#### Steps to reproduce issue
Before my problem happened, `rviz2` was installed via Foxy desktop installation (`sudo apt install ros-foxy-desktop`), and it was working perfectly well. 

Now, I would like to use `tf2` packages as sources (and no longer installed from binaries as it used to be before). So I installed it from sources in my workspace:
`git clone https://github.com/ros2/geometry2.git`
and removed the binaries :
`sudo apt remove '?name(ros-foxy-tf.*)'`
`sudo apt remove '?name(ros-foxy-geometry2.*)'`

Then I built my workspace, and at some point I needed to launch rviz2, with either one of the following command lines:
- `rviz2`
- `ros2 run rviz2 rviz2`

#### Expected behavior
Rviz is supposed to normally launch.

#### Actual behavior
Rviz crashes before being launched. 
Shell display is: 
`[INFO] [rviz2]: Stereo is NOT SUPPORTED`
`[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)`
`[INFO] [rviz2]: Stereo is NOT SUPPORTED`
`rviz2: symbol lookup error: /opt/ros/foxy/lib/librviz_default_plugins.so: undefined symbol: _ZN7tf2_ros6BufferC1ESt10shared_ptrIN6rclcpp5ClockEENSt6chrono8durationIlSt5ratioILl1ELl1000000000EEEE`

#### Additional information
I tried several fixes.
First I removed `geometry2` repo from my workspace and install it again from binaries:
`sudo apt install '?name(ros-foxy-tf.*)'`
`sudo apt install '?name(ros-foxy-geometry2.*)'`

As this fix is not working, I tried to run `rosdep`:
`rosdep update`
`cd <path-to-my-workspace> && rosdep install --from-paths src --ignore-src --rosdistro foxy -y`

Third try, I removed and installed again `rviz`:
`sudo apt remove '?name(ros-foxy-rviz.*)'`
`sudo apt install '?name(ros-foxy-rviz.*)'`

After all these tries, the exact same behavior as described is still happening. So I finally tried to uninstalled and install again ROS2 Foxy, following [ROS Index tutorial](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Debians/):
`sudo apt purge '?name(ros-foxy-.*)' && sudo apt autoremove`
`sudo apt update && sudo apt install curl gnupg2 lsb-release`
`curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -`
`sudo sh -c 'echo "deb [arch=$(dpkg --print-architecture)] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros2-latest.list'`
 `sudo apt update`
`sudo apt install ros-foxy-desktop`
I made sure to source my workspace and Foxy and to run `rosdep`. But, `rviz2` is still crashing at launch time, with the exact same error.

I hope this belongs here as, even if it seems closely related to `tf2` packages. I can move this issue from here to `geometry2` repo if you think it is better. Thanks for helping.

anaelle-sw, 2020-08-25:
Update:

I just solved the problem without intending it, by cleaning and re-building my workspace for another reason. 
Now Rviz2 launches as expected. 

It is a bit surprising to me that cleaning the workspace is the solution, since the dynamic library `/opt/ros/foxy/lib/librviz_default_plugins.so` mentioned in the error message is not located in my workspace. Anyway, I will close this issue.

closed 2020-08-25 by anaelle-sw

eborghi10, 2023-07-31:
@anaelle-sw In my case, RViz2 was crashing with a segmentation fault and I fixed it with this command:

```
sudo apt install --reinstall --yes ros-foxy-tf2-ros
```

I didn't have time to investigate why this is happening.

girishsmurthyIntel, 2024-04-02:
Thank you, I too faced a similar issue with RViz2 crashing with a segmentation fault and the above command helped me resolve.