ros2/rviz #605: Zoom with mouse wheel not working

hdino, 2020-09-13:
In rviz2 version 8.2.0-1focal.20200911.234728 on Ubuntu I cannot zoom with the mouse wheel (default settings). Zooming with a right-click does work though.

I tried to track down this error, but when I compile rviz2, it links to my installation in `/opt/ros/foxy`.

Phymin, 2020-09-30:
I have also encounter this problem in ros2 foxy

paulerikf, 2020-10-16:
Seems like the "Interact" mode is acting just like the "Move camera" mode again?

hdino, 2020-10-16:
True, for me both have the same behaviour.

jacobperron, 2020-10-16:
@paulerikf @hdino Can you confirm if the the interact vs move camera modes worked correctly in a previous ROS 2 version?

jacobperron, 2020-10-17:
@paulerikf Thanks for checking. I think the behavior you see is correct; if the "interact" tool is not hovering over an interactive element (e.g. an interactive marker), then it falls back to camera controls. The same is true in ROS 1 I'm pretty sure.

It sounds like the zooming with the mousewheel has regressed in Foxy.

PR #623 (2020-11-13): Fix for mousewheel to zoom in/out
Fixes #605
closed 2020-11-13 by jacobperron