ros2/rviz #612: rviz2 crashes when using navigation2 in ros2-foxy under ubuntu20.04

Phymin, 2020-10-19:
Hi,

This crash happens when I launch navigation2 with rviz2 in a simulation environment with gazebo11. The following is the error message:
```
[rviz2-1] [INFO] [1603075877.999513614] [rviz2]: Message Filter dropping message: frame 'odom' at time 5078.105 for reason 'Unknown'
[ERROR] [rviz2-1]: process has died [pid 129823, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/phymin/elfin_ros2_ws/install/elfin_navigation/share/elfin_navigation/rviz/elfin_navigation2_view.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_5_ymc7f8'].
```
Sometimes the crash happens when the vehicle nearly reaches the goal, sometimes happens when the vehicle is moving towards the goal, and sometimes happens with no operations, just leave the rviz2 there and it crashes.

At first, I thought maybe there is something wrong with the navigation2, so I remove the binary navigation2 packages and build the navigation2 from source, but the crashes still occurs. So I come here for some help.

The version of my rviz2 versioin is: 8.2.0-1focal.20201008.013454, and I install ros2-foxy from debian package. And the rviz2 configure file is from navigation2 package. Thanks for advance.

jacobperron, 2020-10-28:
Possibly a duplicate of #575. @Phymin Let me know if you think I'm mistaken.

Phymin, 2020-10-28:
Yes, I could be the same problem, so I will close my issue, thank you!

closed 2020-10-28 by Phymin