ros2/rviz #616: rviz2 displays visualization_msgs::msg::Marker at wrong(opposite) position

francis0407, 2020-10-22:
OS: Ubuntu 18.04
ROS: dashing

I'm trying to display a marker (visualization_msgs::msg::Marker) in rviz2. But rviz2 displays the marker at the wrong position.
For example, if I set the point's position as `(1, 1, 0)`, it shows at `(-1, -1, 0)` on rviz2.

Codes for reproducing the problem:
```cpp
#include "rclcpp/rclcpp.hpp"
#include <visualization_msgs/msg/marker_array.hpp>
#include <unistd.h>
using namespace std::chrono_literals;

int main(int argc, char * argv[])
{
    rclcpp::init(argc, argv);
    auto node = rclcpp::Node::make_shared("test_rviz");

    auto pub = node->create_publisher<visualization_msgs::msg::Marker>("test_marker", rclcpp::QoS(1));
    while (!pub->get_subscription_count()) {
        sleep(1);
        std::cout << "sleep" << std::endl;
    }
    std::cout << "pub ->" << std::endl;
    visualization_msgs::msg::Marker marker;
    marker.header.stamp.sec = 0;
    marker.header.stamp.nanosec = 0;
    marker.header.frame_id="world";
    marker.ns = "while_line";
    marker.id = 0;
    marker.type = visualization_msgs::msg::Marker::LINE_LIST;
    marker.action = 0;
    marker.pose = geometry_msgs::msg::Pose();
    marker.scale.set__x(0.08);
    marker.color.a = 1.0;
    marker.color.b = 1.0;
    marker.color.g = 1.0;
    marker.color.r = 1.0;
    auto point1 = geometry_msgs::msg::Point();
    point1.x = 1.0;
    point1.y = 1.0;
    point1.z = 0.0;
    auto point2 = geometry_msgs::msg::Point();
    point2.x = 2.0;
    point2.y = 2.0;
    point2.z = 0.0;
    marker.points.push_back(point1);
    marker.points.push_back(point2);
    pub->publish(marker);
    rclcpp::spin(node);
    rclcpp::shutdown();
    return 0;
}
```

Result on rviz2:
[image]

Result on rviz1 (melodic):
[image]

jacobperron, 2020-10-27:
Interesting. I can reproduce the issue with Dashing on Ubuntu 18.04, but not with Eloquent on Ubuntu 20.04 or Rolling on Ubuntu 20.04.
I'm not sure what changed since Dashing that would have resolved the issue. I'll see if I can identify the change.

jacobperron, 2020-11-21:
I haven't been able to determine why this is a non-issue in Eloquent or newer. I've tried applying the following patches to Dashing but still see the same buggy behavior:

- #375
- #422
- #472
- #394
- #426

paulerikf, 2021-11-23:
Quick note: I have vague memories of seeing a similar issue and needing to explicitly set the marker.pose.orientation.w = 1.0

Seems like the Quaternion message has no default values in [Dashing](https://github.com/ros2/common_interfaces/blob/30dbc60b45e22f2c88a4c46493388fdad3916845/geometry_msgs/msg/Quaternion.msg#L6), whereas it does since [Eloquent](https://github.com/ros2/common_interfaces/blob/81663c07b93889c3d0afda9b99cd5f1c7c98c1f2/geometry_msgs/msg/Quaternion.msg#L6)

jacobperron, 2021-11-23:
@paulerikf ah, that could have something to do with it.

However, since Dashing is EOL (and this isn't an issue in currently supported versions) I'm going to close this ticket.

closed 2021-11-23 by jacobperron

beizhengren, 2022-10-07:
Hi, I meet a error that "fatal error: visualization_msgs/msg/marker_array.hpp: No such file or directory".
I think it's probably wrong with missing some dependencies in cmake. How could I resolve it? 
Thanks.