ros2/rviz #621: Any release plan for the packages in this repository

JaehyunShim, 2020-11-03:
Hi, 

As the title says, I wonder when those packages in the repository will be released, desirably for the foxy version. I am working on developing rviz plugins and have to download and build this repository together every time.  

Thank you in advance,
Jaehyun

jacobperron, 2020-11-03:
For Foxy, we release whatever is on the [foxy branch](https://github.com/ros2/rviz/commits/foxy) (most have already been released). Can you point out particular changes on the ros2 branch that you want to see in Foxy? Note, we typically don't backport features or ABI breaking changes.

JaehyunShim, 2020-11-04:
@jacobperron 

I just ran $ apt update and could install the packages. I checked the ROS index website and they seem to have been released quite recently. Thank you very much.

Jaehyun

closed 2020-11-04 by JaehyunShim