ros2/rviz #627: RViz2 not using configuration in python launch file

martimorta, 2020-11-15:
Hi,
I was calling RViz2 from a launch file and it didn't pick up the configuration file, the arguments kwargs was:

    arguments=['-d '+str(rviz_path)]

I found out that removing the space after d solved the issue

    arguments=['-d'+str(rviz_path)]

The configuration of the system is Ubuntu 20.04 and ROS2 Foxy.

jacobperron, 2020-11-18:
@martimorta can you provide a more complete launch example? Thanks.

martimorta, 2020-11-21:
Sure, this is the fixed one

```
def generate_launch_description():

    base_path = os.path.realpath(get_package_share_directory('package')) # also tried without realpath
    rviz_path=base_path+'/config/config.rviz'

    launch_description = LaunchDescription()
    launch_description.add_action(
        Node(
          package='rviz2', executable='rviz2', name="rviz2", output='screen', arguments=['-d'+str(rviz_path)]
        ),
    )

    return launch_description
```

jacobperron, 2020-12-02:
@martimorta Thanks.

```
arguments=['-d '+str(rviz_path)]
```
does not work because `launch` expects a *list* of arguments, not a single string with spaces between arguments. I think this is a quirk of the underlying asyncio calls in `launch`. I'm pretty sure this has nothing to do with RViz.

I suggest passing your arguments like so:

```
arguments=['-d', str(rviz_path)]
```

closed 2020-12-02 by jacobperron

martimorta, 2020-12-02:
Ah true, thanks!