ros2/rviz #639: Add configuration for Image to be shown without proper TF/fixed frame

RemiRigal, 2021-01-25:
I think that it would be nice to be able to see an image with the `Image` tool even when the `frame_id` provided by the image header is not found in the TF tree or does not match the `fixed frame` in Rviz.

It could maybe just be a checkbox to enable in the `Image` tool ?

bergercookie, 2021-08-10:
I think this is going to be an issue especially with Visual SLAM algorithms, e.g., [orbslam](https://github.com/abhineet123/ORB_SLAM2). When a vSLAM algorithm loses tracking, e.g., due to poor visual information, rapid motion etc. they will stop publishing updates for their corresponding frame until they relocalise / regain tracking. During that duration where tracking is lost, users of rviz wouldn't see any images whatsoever because there wouldn't be a valid link of the frame to the rest of the TF tree.

* Out of cusiosity, what's the rationale behind this? Why *not* show images that don't have an associated `frame`? rviz1 was happy to render images regardless of their frame.
* Roughly what files / what changes would need to be made to show images without an associated frame?

jacobperron, 2021-09-01:
> Out of cusiosity, what's the rationale behind this? Why not show images that don't have an associated frame? rviz1 was happy to render images regardless of their frame.

TBH, I'm not sure this was a conscious decision. It may be more of a bug. It makes sense to me to ignore the fixed from for the image display.

Other displays actually check if the transform exists before deciding to render anything, e.g.

https://github.com/ros2/rviz/blob/b30097c0918eb8e381d96a2178f39b8acdddf4f5/rviz_default_plugins/src/rviz_default_plugins/displays/odometry/odometry_display.cpp#L276

However, the ImageDisplay does not do such a check, so I'm not sure why images are not being rendered...:

https://github.com/ros2/rviz/blob/b30097c0918eb8e381d96a2178f39b8acdddf4f5/rviz_default_plugins/src/rviz_default_plugins/displays/image/image_display.cpp#L195-L207

Maybe it's something to do with the texture?

helenol, 2021-10-05:
+1 to this topic, blocking development for me and a huge regression from rviz1. Why would I need my whole TF tree to be resolved correctly to see 2D images?

helenol, 2021-10-05:
It appears to be because of the TF filter on incoming image messages? https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/include/rviz_default_plugins/displays/image/image_transport_display.hpp#L117-L140 So this is explicitly added? Why?

andreasBihlmaier, 2021-12-26:
+1 on changing this behavior back to ROS1 behavior: Show images no matter what is available in TF.

At the very least, do not show "(checkmark-symbol) Status: OK", when it should be "(error-symbol) Status: Missing TF _<tf-frame-in-image>_ transform".

PR #788 (2022-01-13): Remove TF filter from ImageTransportDisplay
Fixes #639

Only the CameraDisplay cares about TF data, so this change moves the TF filter logic into the CameraDisplay.

This allows users to view images with the ImageDisplay no matter what the fixed frame is set to.
commit f8e8163a (2022-01-13): Remove TF filter from ImageTransportDisplay (#788)

* Remove TF filter from ImageTransportDisplay

Fixes #639

Only the CameraDisplay cares about TF data, so this change moves the TF filter logic into the CameraDisplay.

This allows users to view images with the ImageDisplay no matter what the fixed frame is set to.
closed 2022-01-13 by jacobperron