ros2/rviz #640: Python API

BasVolkers, 2021-01-29:
Hi, im looking to migrate my package to ROS2. 
This package uses the python_bindings of rviz in ROS1.
Are these also available somewhere for ROS2? I can't seem to find them.

clalancette, 2021-02-18:
We don't currently have Python bindings available for the ROS 2 version of RViz.  I've assigned the issue to one of the rviz maintainers in ROS 2 to take a look at the feasibility of doing so.  Likely this will need community contribution to do so, as it isn't on our current roadmap.

jacobperron, 2021-02-23:
I don't know much about the Python bindings in ROS 1, but it [looks like](https://github.com/ros-visualization/rviz/tree/882ee38782517e60a69734c471eb9c09c208088a/src/python_bindings) we are generating the Python API from the C++ code with [shiboken](https://wiki.qt.io/Qt_for_Python/Shiboken) and/or [sip](https://pypi.org/project/sip/). So I think doing something similar might be feasible for ROS 2.

For anyone who is willing to tackle this, I'd rather see shiboken used for Python API generation over sip. This is mainly because of licensing (shiboken is LGPL, while sip is GPL). Furthermore, IIUC, Qt has officially adopted shiboken as well. It looks like the [shiboken generator was disabled a long time ago in ROS 1](https://github.com/ros-visualization/rviz/blob/882ee38782517e60a69734c471eb9c09c208088a/src/python_bindings/CMakeLists.txt#L3-L4), but I guess it doesn't mean much for ROS 2, since the C++ code is very different.

Another option for exposing a Python API could be to use [pybind11](https://pybind11.readthedocs.io/en/stable/). FWIW, we are using pybind11 for rosbag2 Python bindings and are [currently in progress of refactoring rclpy to use pybind11](https://github.com/ros2/rclpy/issues/665).

This needs someone to take a deeper look, but I'd expect we might have an easier time using shiboken considering it actively being used by other Qt-based projects.

EricCousineau-TRI, 2021-03-09:
See also: https://gitlab.kitware.com/autopybind11/autopybind11/-/blob/55f7b955a65622414ebd961325779e3c404c3e13/README.rst#alternatives-comparison

ipa-cmh, 2022-08-24:
I have made a first attempt with shiboken here: https://github.com/ipa-cmh/rviz/tree/add-python-bindings
It builds... Still some work and testing to though. Will open a PR when I am a little further along.

friackazoid, 2023-02-09:
Hello, I want to port one python software to ROS2 and that is required python bindings for RVIZ. Can I help with this PR?  what is required for the merge?

ipa-cmh, 2023-02-10:
@friackazoid there is #889 which works under circumstances. There is still a build dependency problem which I have only been able to solve for local build with python_qt_binding in same workspace. Also to merge Windows support is required. I have no idea where to start and test that.

I have a more recent version that we use in some internal projects - still throws a lot of warnings durign build, but works. Honestly, my funding for this activity got scrapped, so I cannot pursue it further right now. I will try and push the latest working version I have to #889.

Also check out https://github.com/ros-visualization/python_qt_binding/issues/114 which documents my effort to update python_qt_bindings.

KKSTB, 2023-08-11:
Thanks for your effort @ipa-cmh.

I would like to try VisualizationFramePy in my program. I downloaded the latest PR: https://github.com/ipa-cmh/rviz/tree/f324c63a5035c2351ddf599fe75e1995119ea89c

I tried to build it in this environment:
OS: Ububtu 22.04
ROS: iron
Build command: colcon build --merge-install --allow-overriding rviz2 rviz_default_plugins

But I am stuck at this colcon_cmake exception:
>--- stderr: rviz_python_bindings                                                                                              
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_cmake/build.py", line 59, in build
    rc = await extension.build(
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 182, in _reconfigure
    buildfile = get_buildfile(cmake_cache)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 112, in get_buildfile
    if 'Ninja' in generator:
TypeError: argument of type 'NoneType' is not iterable

At `colcon_cmake/task/cmake/build.py", line 182`, it seems there is some sort of build environment reconfigure failure causing a non-zero return code (as indicated by the comments in colcon_cmake/task/cmake/build.py).

I have installed python3_pyside2 and python3_qtpy (though not by rosdep, but by apt install `python3-qtpy libpyside2-dev libshiboken2-dev python3-pyside2.qtcore python3-pyside2.qtgui python3-pyside2.qthelp python3-pyside2.qtnetwork python3-pyside2.qtprintsupport python3-pyside2.qtsvg python3-pyside2.qttest python3-pyside2.qtwidgets python3-pyside2.qtxml shiboken2` as indicated in https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml

Not sure what build environment is further required. Thanks.

KKSTB, 2023-08-14:
I tried hard locating the problem using the verbose build command:
`VERBOSE=1 colcon build --merge-install --allow-overriding rviz2 rviz_default_plugins --event-handlers console_cohesion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON`

and is able to locate the exception location (https://github.com/ipa-cmh/rviz/blob/f324c63a5035c2351ddf599fe75e1995119ea89c/rviz_python_bindings/CMakeLists.txt#L50).

I substituted `ament_get_recursive_properties()` with older version of include_dirs method in rviz_python_bindings CMakeLists.txt instead (https://github.com/ipa-cmh/rviz/blob/05f5a746b472aeadf2d58bb2c51b2198a8327b96/rviz_python_bindings/CMakeLists.txt#L83)

However, there is still a missing cmake function `shiboken_generator_ext()` at https://github.com/ipa-cmh/rviz/blob/f324c63a5035c2351ddf599fe75e1995119ea89c/rviz_python_bindings/CMakeLists.txt#L82.

Even though I try to replace it with `shiboken_generator()` of https://github.com/ros-visualization/python_qt_binding/blob/rolling/cmake/shiboken_helper.cmake, there is still a lot of build errors that I dont know how to proceed yet.

ipa-cmh, 2023-08-15:
@KKSTB sorry mate, I was on vacation and it has been some time I worked on this, currently the funding has gone away again.
I think I have a version somewhere that will build directly, but it is not very nicely done and still produces a lot of errors.

In general, I had quite some problems with making this build stably, because the build cache and install folders usually contain some stuff that will make you miss some errors when doing changes.

In the end, we are not using rviz2 for end-user interfaces anymore. We now switched to nicegui.
https://github.com/zauberzeug/nicegui

KKSTB, 2023-08-16:
Hi nice to hear from you @ipa-cmh.

I try to use the successfully built rviz_python_bindings in a freshly created ROS2 python package with an empty pyside window. It displays an empty window as intended, but it crashes with segmentation fault once I add a line to import rviz_shiboken:
```
Current thread 0x00007fc09e05e740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1078 in _handle_fromlist
  File "/home/kk/.local/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142 in _import
...
```

Hope that there would be help on how to solve this issue.

ipa-cmh, 2023-08-16:
Try the following for import, that is how I have it in our project.

```python
import qtpy
from qtpy.QtGui import *
from qtpy.QtCore import *

try:
    from qtpy.QtWidgets import *
except ImportError:
    pass
from rviz import YamlConfigReader, VisualizationFrame, Config, rviz_common

```

You can than do the following:

```python
with open("parameters.yml", "w") as outfile:
  yaml.dump(param_file, outfile, default_flow_style=False)
frame = VisualizationFrame(str(Path("parameters.yml").absolute()))
frame.setApp(app)
frame.setSplashPath("")
frame.initialize()
sleep(3.0)
frame.loadDisplayConfig(file)
```

Finally you'll have to add the frame to your layout.

KKSTB, 2023-08-16:
Thanks for your quick reply. I found that it is my system installation problem.

I found that if I use Qt classes in `python_qt_binding` that all rqt plugins are using, I still have to `pip3 install PySide2` even though I have apt installed the pyside2 components.

If I freshly create ROS2 python package without following rqt plugin, and use `qtpy` just as your example did, then the pip3 PySide2 is not required.

After pip uninstalling pyside2 there is no crash problem. Will continue trying. Thanks @ipa-cmh!

KKSTB, 2023-08-17:
Your example is very essential. I found that I have to call `frame.setApp(app)` in order to run it (where app is a QApplication), otherwise it crashes with code -11. ROS1 version of `VisualizationFrame` does not need this to run.

I think the major trouble is the difficulty to debug the issue. It often crashes with exit code -11 or causes segmentation fault without indicating the reason if configuration is not followed strictly (e.g. 1: pyside compatibility problem with `python_qt_binding`. 2: the need to call `frame.setApp(app)`).

KKSTB, 2023-08-17:
There is yet another problem. Even though I can display the VisualizationFrame using my ROS2 package, the 3D screen is flashing. If I `ros2 run rviz2 rviz2` directly there is no flashing problem.

Minimal code of my ROS2 package:
```python
import os
import sys
import rviz

from qtpy.QtWidgets import QApplication

from ament_index_python.packages import get_package_share_directory

def main():
    app = QApplication(sys.argv)
    rviz_frame = rviz.VisualizationFrame()
    rviz_frame.setApp(app)
    rviz_frame.setSplashPath("")
    rviz_frame.initialize(os.path.join(get_package_share_directory('test_gui'), 'resource', 'default.rviz'))
    rviz_frame.show()
    sys.exit(app.exec_())

if __name__ == '__main__':
    main()
```

ipa-cmh, 2023-08-17:
> Your example is very essential. I found that I have to call `frame.setApp(app)` in order to run it (where app is a QApplication), otherwise it crashes with code -11. ROS1 version of `VisualizationFrame` does not need this to run.

This is caused by an API change of the underlying c++-Code. There is also the issue that you have to pass in a smart node pointer to the constructor of VisualizationFrame now. Therefore, there is now some helper code necessary.

> I think the major trouble is the difficulty to debug the issue. It often crashes with exit code -11 or causes segmentation fault without indicating the reason if configuration is not followed [...]

Overall the whole wrapping part is pretty complex and if you have pyside and pyqt installed there is also often confusion about the version that is used. Generally, I would only use and work on the wrapper if you really need it. Otherwise stick with C++ for now.

KKSTB, 2023-08-18:
Python is much easier and much faster to develop for average user. Also ROS1 has rviz python binding, so its normal to expect ROS2 to have it too.

I tried to embed a C++ version of `rviz_common::VisualizationFrame`. Unfortunately it seems to work fine without flashing.
```C++
#include <qt5/QtWidgets/QApplication>
#include <qt5/QtWidgets/QMainWindow>

#include "rclcpp/rclcpp.hpp"

#include <rviz_common/visualization_frame.hpp>
#include "rviz_common/ros_integration/ros_client_abstraction.hpp"
#include "rviz_common/ros_integration/ros_node_abstraction.hpp"

int main(int argc, char* argv[])
{
//  rclcpp::init(argc, argv);

//  RCLCPP_INFO(rclcpp::get_logger("my_logger"), "start");

  QApplication app(argc, argv);

  QMainWindow window = QMainWindow();

  std::unique_ptr<rviz_common::ros_integration::RosClientAbstraction> ros_client_abstraction_ = std::make_unique<rviz_common::ros_integration::RosClientAbstraction>();
  rviz_common::ros_integration::RosNodeAbstractionIface::WeakPtr rviz_ros_node_ = ros_client_abstraction_->init(argc, argv, "rviz", false);
  rviz_common::VisualizationFrame rviz_frame(rviz_ros_node_);
  rviz_frame.setApp(&app);
  rviz_frame.setSplashPath("");
  rviz_frame.initialize(rviz_ros_node_);

  window.setCentralWidget(&rviz_frame);
  window.show();

  RCLCPP_INFO(rclcpp::get_logger("my_logger"), "exec");

  int ret = app.exec();

  ros_client_abstraction_->shutdown();

//  rclcpp::shutdown();

  return ret;
}
```

I also modified `visualizer_frame_py.cpp` so that `rviz_python_bindings` can be built without having to modify rviz2, and built with ROS2 iron without rviz source. Unfortunately it doesnt fix the flashing issue.
```C++
#include "rviz_python_bindings/visualizer_frame_py.hpp"

using namespace rviz_common;

VisualizerFramePy::VisualizerFramePy(QWidget * parent)
: VisualizationFrame(rviz_common::ros_integration::RosNodeAbstractionIface::WeakPtr(), parent)
{
  ros_client_abstraction_ = std::make_unique<rviz_common::ros_integration::RosClientAbstraction>();
  rviz_ros_node_ = ros_client_abstraction_->init(0, nullptr, "rviz", false);
}

VisualizerFramePy::~VisualizerFramePy()
{
  ros_client_abstraction_->shutdown();
}

bool VisualizerFramePy::node_ok()
{
  return ros_client_abstraction_->ok();
}

void VisualizerFramePy::initialize(
  const QString & display_config_file)
{
  VisualizationFrame::initialize(rviz_ros_node_, display_config_file);
}
```

So I think the problem lies in python / pyside / shiboken / display.

KKSTB, 2023-08-21:
I found that in pyside environment, `ogre_render_window_->windowMovedOrResized()`(https://github.com/ros2/rviz/blob/7a2a6bc2e9cd8df996d7f320ef8c166638691d47/rviz_rendering/src/rviz_rendering/ogre_render_window_impl.cpp#L273C9-L273C23) raises `RenderWindow::exposeEvent()`(https://github.com/ros2/rviz/blob/7a2a6bc2e9cd8df996d7f320ef8c166638691d47/rviz_rendering/src/rviz_rendering/render_window.cpp#L191C5-L191C5), which in turn calls `RenderWindowImpl::resize()`, resulting in infinite loop of events causing flash problem.

Whereas in C++ environment, `ogre_render_window_->windowMovedOrResized()` does not raise `RenderWindow::exposeEvent()`.

If I comment either of the above lines, then there is no flash, but resizing becomes not properly handled.

KKSTB, 2023-08-23:
The window size problem is related to the if condition inside resize event. After completely removing `isExposed()` condition and `exposeEvent()` there is no flash problem and initial size problem.

KKSTB, 2023-08-29:
I ran shiboken2.exe directly to look into the problem. It was due to compatibility with `MSVC v142/latest` (https://github.com/KDAB/KDDockWidgets/issues/114).

After installing `MSVC v142/v14.26` on top of v142/latest and adding `-vcvars_ver=14.26` in the `x64 Native Tools Command Prompt` shortcut, there is some output. It will then ask for something called `console_bridge` (https://github.com/ros/console_bridge).

Finally shiboken2.exe crashes with no output for any #include involving rclcpp / any #include inside rclcpp. It is difficult to proceed any further with no output.

So unless rviz adds a wrapper class / abstraction interface to completely hide rclcpp, there is little hope for rviz to support shiboken python binding on Win10 (i.e. only Ubuntu works).

vamsi1961, 2024-09-15:
I want to create a UI using Qt5 and integrate ROS 2 visualization into it. How can I achieve this?