ros2/rviz #645: Incorporating RViz2 into custom GUI (Qt app)

Abishalini, 2021-02-04:
I am trying to port librviz to ROS2/RViz2 ([http://docs.ros.org/en/indigo/api/librviz_tutorial/html/index.html])
I could not find visualization_manager class (I built RViz2 along with ROS2 and I couldn't find the appropriate files for it on local). So I copied over visualization_manager.cpp/hpp into my workspace and linked it to `rviz_common::rviz_common` in CMakeLists.txt

Doing so gave me the following errors -
`[rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.1/src/rviz_rendering/render_system.cpp:259
[rviz_rendering:info] Stereo is NOT SUPPORTED, at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.1/src/rviz_rendering/render_system.cpp:508
[rviz_rendering:info] OpenGl version: 4.6 (GLSL 4.6), at /tmp/binarydeb/ros-foxy-rviz-rendering-8.2.1/src/rviz_rendering/render_system.cpp:242
[rviz_common:debug] Load pixmap at package://rviz_default_plugins/icons/classes/TF.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_default_plugins/icons/classes/TF.svg]: Couldn't open file /opt/ros/foxy/share/rviz_default_plugins/icons/classes/TF.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_default_plugins/icons/classes/TF.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Load pixmap at package://rviz_common/icons/classes/Identity.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_common/icons/classes/Identity.svg]: Couldn't open file /opt/ros/foxy/share/rviz_common/icons/classes/Identity.svg, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_common/icons/classes/Identity.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:71
[rviz_common:debug] Error retrieving file [file:///opt/ros/foxy/share/rviz_common/icons/classes/Identity.png]: Couldn't open file /opt/ros/foxy/share/rviz_common/icons/classes/Identity.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:55
[rviz_common:debug] Load pixmap at package://rviz_common/icons/default_class_icon.png, at /tmp/binarydeb/ros-foxy-rviz-common-8.2.1/src/rviz_common/load_resource.cpp:71`

when I tried to make an instance of the visualization_manager class
`rviz_common::ros_integration::RosNodeAbstractionIface::WeakPtr ros_node_abstraction_ = std::weak_ptr<rviz_common::ros_integration::RosNodeAbstractionIface>();
  rviz_common::WindowManagerInterface* wm_ = nullptr;
  rclcpp::Clock::SharedPtr clock_ = std::make_shared<rclcpp::Clock>(RCL_SYSTEM_TIME);

  manager_ = new rviz_common::VisualizationManager( render_panel_ , ros_node_abstraction_, wm_,  clock_);`

How do I go about this? The goal is to incorporate Rviz into a custom Qt App

schornakj, 2021-02-04:
I'm also very interested in figuring out how to do this, since embedded Rviz displays are a useful feature to have in nicely-packaged custom robotics applications.

schornakj, 2021-02-07:
I wanted to provide some detailed information about what I'm trying to do and my current issues. I've been referring to t[he functions used while initializing the rviz2 application](https://github.com/ros2/rviz/blob/c5ff87d4fc92856ec901e45403ae33db8e8281ab/rviz_common/src/rviz_common/visualization_frame.cpp#L245-L374) while trying to get this working. This is also building from the code that @Abishalini wrote while originally working on this issue.

I created [a branch of ros2/rviz](https://github.com/schornakj/rviz/tree/feature/expose-visualization-manager) that moves the `VisualizationManager` to the public API, so that's what I'm linking against in the code below. Based on [examples from ROS1](http://docs.ros.org/en/indigo/api/librviz_tutorial/html/index.html) for how to set up a custom application like this I think this change is needed to achieve my desired result, but since my application doesn't work yet that's a speculative assumption.

Basically I would like to have an app that looks something like this (minus the sliders and labels -- I deleted those from my code for clarity):
[image]

Please let me know if I should provide additional information. One of the motivations behind this is to be able to port the MoveIt Setup Assistant to ROS2, since the ROS1 version of that application follows a similar pattern.

## Code
### myviz.h
```
#pragma once

#include <QWidget>
#include <QMainWindow>
#include <rclcpp/rclcpp.hpp>

namespace rviz_common
{
class Display;
class RenderPanel;
class VisualizationManager;
}

class MyViz: public QMainWindow, public rclcpp::Node
{
Q_OBJECT
public:
  MyViz( QWidget * parent = 0 );

private:
  rviz_common::VisualizationManager* manager_;
  rviz_common::RenderPanel* render_panel_;
  rviz_common::Display* grid_;
};
```

### myviz.cpp
```
#include "myviz.h"

#include <QColor>
#include <QGridLayout>
#include <QVBoxLayout>

#include "rclcpp/clock.hpp"
#include "rviz_common/render_panel.hpp"
#include "rviz_common/display.hpp"
#include <rviz_common/display_context.hpp>
#include "rviz_common/ros_integration/ros_node_abstraction_iface.hpp"
#include "rviz_common/ros_integration/ros_client_abstraction.hpp"
#include "rviz_common/ros_integration/ros_node_abstraction.hpp"
#include "rviz_common/visualization_manager.hpp"
#include "rviz_rendering/render_window.hpp"

MyViz::MyViz( QWidget* parent )
  : QMainWindow( parent )
  , Node("rviz_gui")

{
  QWidget * center_widget = new QWidget(this);

  auto ros_node_abs = std::make_shared<rviz_common::ros_integration::RosNodeAbstraction>("rviz_render_node");
  rviz_common::WindowManagerInterface* wm_ = nullptr;
  auto clock_ = std::make_shared<rclcpp::Clock>(RCL_SYSTEM_TIME);

  render_panel_ = new rviz_common::RenderPanel();
  manager_ = new rviz_common::VisualizationManager( render_panel_ , ros_node_abs, wm_,  clock_);

[... 119 lines cut ...]

### Stack Trace
```
1  __GI_raise                                                 raise.c                     50   0x7fa29d29f18b 
2  __GI_abort                                                 abort.c                     79   0x7fa29d27e859 
3  __libc_message                                             libc_fatal.c                155  0x7fa29d2e93ee 
4  malloc_printerr                                            malloc.c                    5347 0x7fa29d2f147c 
5  _int_free                                                  malloc.c                    4177 0x7fa29d2f2cbc 
6  XFree                                                                                       0x7fa29a70554d 
7  Ogre::GLXWindow::windowMovedOrResized                      OgreGLXWindow.cpp           694  0x7fa2801f2e2d 
8  rviz_rendering::RenderWindowImpl::resize                   ogre_render_window_impl.cpp 267  0x7fa29c62ddc5 
9  rviz_rendering::RenderWindow::windowMovedOrResized         render_window.cpp           123  0x7fa29c63925c 
10 rviz_common::RenderPanel::resizeEvent                      render_panel.cpp            300  0x7fa29e7cbac4 
11 QWidget::event(QEvent *)                                                                    0x7fa29dfa3947 
12 QApplicationPrivate::notify_helper(QObject *, QEvent *)                                     0x7fa29df60a66 
13 QApplication::notify(QObject *, QEvent *)                                                   0x7fa29df6a0f0 
14 QCoreApplication::notifyInternal2(QObject *, QEvent *)                                      0x7fa29d8cf93a 
15 QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool)                                  0x7fa29df9be6e 
16 QWidgetPrivate::show_helper()                                                               0x7fa29df9fbd7 
17 QWidgetPrivate::setVisible(bool)                                                            0x7fa29dfa2d6b 
18 QWidgetPrivate::showChildren(bool)                                                          0x7fa29df9fb61 
19 QWidgetPrivate::show_helper()                                                               0x7fa29df9fbf3 
20 QWidgetPrivate::setVisible(bool)                                                            0x7fa29dfa2d6b 
21 main                                                       main.cpp                    17   0x559e6196fba4 
```

schornakj, 2021-02-08:
It's possibly worth noting that I also get errors about failures to load certain icon images:
```
[rviz_common:debug] Error retrieving file [file:///home/jschornak/workspaces/moveit2_foxy_debug_ws/install/rviz_default_plugins/share/rviz_default_plugins/icons/classes/MoveCamera.svg]: Couldn't open file /home/jschornak/workspaces/moveit2_foxy_debug_ws/install/rviz_default_plugins/share/rviz_default_plugins/icons/classes/MoveCamera.svg, at /home/jschornak/workspaces/moveit2_foxy_debug_ws/src/rviz/rviz_common/src/rviz_common/load_resource.cpp:55

```
In these cases the image `MoveCamera.svg` does not exist at the requested path.

clalancette, 2021-02-25:
Assigning to @jacobperron to take a look.  Most likely this results in this being marked as 'help-wanted' or 'backlog'.  Contributions here would be welcome.

schornakj, 2021-02-25:
@clalancette We were able to work through this and get our application working. I'll find a good place to post a minimal open-source example based on that `librviz` tutorial. Some changes to the core RViz packages were required, and I submitted #649 adding these.

clalancette, 2021-02-25:
Great, thanks for the heads up.

JaehyunShim, 2021-04-06:
@schornakj  Did it work for you guys? I downloaded the source code and tried it but it dies when calling the [this function](https://github.com/ros2/rviz/blob/ros2/rviz_common/src/rviz_common/visualization_manager.cpp#L222)

Thank you in advance
Jaehyun

clalancette, 2021-04-06:
Closing this as #649 was merged.

closed 2021-04-06 by clalancette

SEUZTh, 2023-07-06:
> 

Hi, have you solve the problem?

vamsi1961, 2024-09-15:
@schornakj  how did you integrate rvz2 vizualization panel in qt5 ui

xingzay, 2024-12-19:
> @schornakj how did you integrate rvz2 vizualization panel in qt5 ui. I want to integrate in humble

Hello! I also want to implement ros2 humble to integrate rviz2 on qt5. Did you save the problem? I'm not sure if the RVIZ2 API can be applied with QT5

SEUZTh, 2024-12-20:
> > @schornakj how did you integrate rvz2 vizualization panel in qt5 ui. I want to integrate in humble您是如何将 rviz2 可视化面板集成到 Qt5 UI 中的？我想在谦卑中集成。
> 
> Hello! I also want to implement ros2 humble to integrate rviz2 on qt5. Did you save the problem? I'm not sure if the RVIZ2 API can be applied with QT5你好！我也想实现 ros2 humble 以集成 rviz2 到 qt5。你保存了这个问题吗？我不确定 RVIZ2 API 是否可以应用于 QT5

https://github.com/SEUZTh/ros2_qt_template

SEUZTh, 2024-12-20:
> @schornakj how did you integrate rvz2 vizualization panel in qt5 ui. I want to integrate in humble您是如何将 rviz2 可视化面板集成到 Qt5 UI 中的？我想在谦卑中集成。

https://github.com/SEUZTh/ros2_qt_template