ros2/rviz #646: Cannot View Images

bostoncleek, 2021-02-09:
I am trying to view an image (sensor_msgs/msg/CameraInfo). The data is being published after echoing the topic. 
[image]

When running rviz I get the following message:
```
qt5ct: using qt5ct plugin
[INFO] [1612912407.586356395] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1612912407.586447785] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
qt5ct: D-Bus global menu: no
```

mjeronimo, 2021-02-26:
@bostoncleek Assuming that your topics match and the QoS settings also match, the next likely reason is that the sim_time settings don't match between the publisher and rviz. This is relevant as rviz uses a message_filter for its subscriptions and a sim_time mismatch won't allow the messages to pass.

jacobperron, 2021-03-12:
Please also confirm that the frame ID in the camera message also matches the "Fixed Frame" in the RViz global options.

bostoncleek, 2021-03-17:
@jacobperron I set the Fixed Frame to camera_color_optical_frame which is the frame ID on topic /camera/color/image_raw. The image is not available.

mjeronimo, 2021-03-17:
@bostoncleek Did you confirm matching QoS and sim_time settings?

closed 2022-02-03 by clalancette