ros2/rviz #663: rviz2 fails to build on mac os big sur aarch64

YVbakker, 2021-03-15:
when building according to the instructions in this repos README, the build fails when building rviz_ogre_vendor:
```
objc[55948]: Class AMSupportURLConnectionDelegate is implemented in both ?? (0x1f650c188) and ?? (0x1212042b8). One of the two will be used. Which one is undefined.
objc[55948]: Class AMSupportURLSession is implemented in both ?? (0x1f650c1d8) and ?? (0x121204308). One of the two will be used. Which one is undefined.
Undefined symbols for architecture x86_64:
```
see https://gist.github.com/YVbakker/8d943865546469d77633c0a078925c17 for the full colcon output
I will be updating this issue with new revelations as they come

YVbakker, 2021-03-15:
The problems seem to be occurring in:
```
[ 52%] Building CXX object Components/RTShaderSystem/CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderCGProgramProcessor.cpp.o
Undefined symbols for architecture x86_64:
...
[ 53%] Linking CXX shared library ../../lib/macosx/Codec_EXR.dylib
Undefined symbols for architecture x86_64:
...
[ 65%] Linking CXX shared library ../../lib/macosx/libOgreOverlay.dylib
Undefined symbols for architecture x86_64:
```

YVbakker, 2021-03-24:
I identified the problem; the ogre libraries were simply not building for arm64, but some dependencies (such as truetype) were. I updated the ogre build scripts to include the arm64/APPLE combo on top of the intel and powerpc variants. At this point it may be advisable to do a general version bump from ogre version 1.12.1 to 1.12.11 anyway, but the procedure should be nearly identical for either of these versions.

Furthermore, there is an issue with qt5 on arm64, so I had to resort to the just released qt6. This means that I had to make a few changes to the code that deals with qt implementation. I only changed the most dire though, there still is some work to do regarding the *DEPRECATED* qt5->qt6 features. Luckily these are just warnings for now, and I now have rviz2 running natively on mac osx big sur, aarch64. If anyone reading this is interested in the sources, please remind me to make a pull request somewhere as I still have the diff here.

clalancette, 2021-04-08:
@YVbakker Thanks for doing the investigation!  I've assigned this issue to you to help remind you to open a PR with the changes. I'll also mark this as a bug, and put it on our backlog, but we may not get to it anytime soon.

YVbakker, 2021-04-09:
No problem! I haven't updated the issue yet, but in the mean time I have rviz2 running natively on mac os 11 arm64. At the same time, because of a dependency issue with qt5 I was forced to build against qt6 and in the process fix some compatibility issues. So if done right I could potentially contribute qt6 support as well.