ros2/rviz #666: Issue rendering cylinders

DeborggraeveR, 2021-03-21:
When i add cylinders to my robot model, it seems they cannot handle decimal values.
Decimal values do work correctly with boxes.

This line works
<cylinder radius="1" length="1"/>

This line shows a black cylinder (0 height)
<cylinder radius="1" length="0.5"/>

This line become invisible
<cylinder radius="0.5" length="1"/>

These lines show exact same shape
<cylinder radius="1" length="1"/>
<cylinder radius="1" length="1.5"/>

geehalel, 2021-03-29:
It seems to be locale related. Using commas as decimal points in the urdf file solves the issue  (my locale is fr_FR). But that's not a solution. Simply launch rviz2 without a locale defined
```
LANG= rviz2
```
Just starting using ros2, can't help more at the moment. But what a fantastic framework.

clalancette, 2023-07-14:
Oh, this was solved in URDF quite a while ago.

Because of that, I'm going to assume that this is fixed, and close this issue.  If this is not the case, please feel free to reopen.

closed 2023-07-14 by clalancette