ros2/rviz #667: Camera not showing LaserScan

martimorta, 2021-04-05:
Hi,
minor issue here, trying to see a LaserScan on a Camera display, it shows the odometry and robot model but it doesn't show a LaserScan, descriptive screenshot below on a Ubuntu 20.04 and Foxy.
Cheers
[image]

mjeronimo, 2021-04-22:
@martimorta  I tried to reproduce on Foxy using the nav2 stack, but was unable to. Also tried the latest Rolling code. In both cases, I was able to view both the Odometry arrows and the laser scan. I'll need more info/files to reproduce.

martimorta, 2021-04-23:
Hi Michael, just tried today after updates, same code on my side and I can see everything 🤷‍♂️ all good!

closed 2021-04-23 by martimorta

mjeronimo, 2021-04-23:
@martimorta Great, glad to hear it.