ros2/rviz #671: rviz2 creates multiple nodes with the same name when Node name is set

shonigmann, 2021-04-08:
## Bug report
Launching an rviz2 node with the name argument set results in multiple nodes with the same name being spawned. I have only tried one .rviz config file that generates this behavior so far (from nav2_bringup 
[nav2_default_view.rviz.txt](https://github.com/ros2/rviz/files/6281438/nav2_default_view.rviz.txt) for reference, converted to .txt to allow upload), but the behavior seems to be related to how rviz is launched, so I don't think its _only_ a config file issue. 

**Required Info:**

- Operating System:
  - Ubuntu 20.04 
- ROS2 Version:
  - Foxy binaries 
- Version or commit hash:
  - ros-foxy-rviz-assimp-vendor	8.2.1-1focal.20201210.052742
  - ros-foxy-rviz-common	8.2.1-1focal.20210225.225538
  - ros-foxy-rviz-default-plugins	8.2.1-1focal.20210225.235655
  - ros-foxy-rviz-ogre-vendor	8.2.1-1focal.20201210.052821
  - ros-foxy-rviz-rendering	8.2.1-1focal.20210125.200440
  - ros-foxy-rviz2	8.2.1-1focal.20210226.004041

#### Steps to reproduce issue

I can reproduce this issue both using the CLI and using Launch files. 

On the commandline, with the attached file: 
```
ros2 run rviz2 rviz2 -d nav2_default_view.rviz.txt --ros-args --remap __node:=test
```

#### Expected behavior
A single rviz node is created, with the name "test"

#### Actual behavior
4 rviz nodes are created resulting in the following warning:
```
[rviz2-4] [WARN] [1617916304.102991367] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

and `ros2 node list` output:
```
ros2 node list 
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
...
/test
/test
/test
/test
```
#### Additional information
If the node name is not set (e.g. just running `ros2 run rviz2 rviz2 -d nav2_default_view.rviz.txt`), rviz runs as expected without any duplicate nodes. 

```
ros2 node list 
...
/rviz
```

I tried to reproduce this using the default rviz config (e.g. opening rviz without any arguments, saving the config, closing rviz, and rerunning rviz with the config file and a node name set) and only 1 node was created... So the behavior feels at least in part due the config file.

mjeronimo, 2021-04-22:
@shonigmann When launching with 

`ros2 run rviz2 rviz2 -d nav2_default_view.rviz`

There are actually four nodes created:

- **/lifecycle_manager_localization_service_client**
      This is a (non-spinning) node used to call the lifecycle manager service. 
- **/lifecycle_manager_navigation_service_client**
       This is a (non-spinning) node used to call the lifecycle manager service. 
- **/rviz**
       The main rviz node.
- **/transform_listener_impl_\<n\>**
       A node created by the transform listener. 

If you use: `--ros-args --remap __node:=test`, you're requesting that every node be renamed to test, which is what happens. 

Instead, you can target the nodes individually, if you like:

`rviz2 --display-config nav2_default_view.rviz --ros-args --remap rviz:__name:=test --remap lifecycle_manager_localization_service_client:__name:=lc_mgr_client`

Check this page out for more details: https://design.ros2.org/articles/static_remapping.html

shonigmann, 2021-04-22:
@mjeronimo, that makes sense, thanks for the clear explanation! I'll see if I can get a pull request in over on the nav2 stack to implement the more explicit node naming to avoid the warning message. Cheers

closed 2021-04-22 by shonigmann