ros2/rviz #672: RenderingAPIException causes rviz2 to crash after 100 tries.

jasmeet0915, 2021-04-15:
Hi, 
I am not able to get rviz2 to start on my system due to the following error: 

`[ERROR] [1618426295.882467513] [rviz2]: RenderingAPIException: Unable to create a suitable GLXContext in GLXContext::GLXContext at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.1/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXContext.cpp (line 60)`

I have ubuntu 20.04 with ros2 foxy and ignition gazebo installed on my system both of which are running fine.

From what I understood, I thought this was a version issue. I have libogre versions 1.8, 1.9 and 2.1 which were installed by the ignition-rendering libraries.  But according to the above error message, I guess rviz2 is looking for ogre-1.12 which is not installed. I tried installing the `libogre-1.12` and `libogre-1.12-dev` packages from apt but still no luck and installing this package also removes the `ignition-citadel `package which I have installed.

Does anyone have idea how to resolve this issue?

Cheers
Jasmeet

david-beaudette, 2021-04-20:
Also had this problem, unable to start rviz2 after the latest Ubuntu upgrade of foxy. I was able to run rviz2 before without issues. A simple reboot fixed the issue on my side.

jasmeet0915, 2021-04-21:
@david-beaudette ohh. I have tried rebooting and upgrading but still it doesn't work. I am facing this issue from the beginning. I have not been able to start rviz2 since I installed it. 

Also I Just noticed this was also printed in the beginning before the 100 lines of the above error:
`libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
`
Tried some solutions for this which I found online but still it is crashing.

gbalke, 2021-09-22:
@jasmeet0915 I found this discussion that appears to have a similar issue: https://github.com/openai/gym/issues/509#issuecomment-282331680

It seems most likely that you're missing this driver? I see that you've already tried some solutions online so please let me know if you've already tried this!

Also, are you running in headless mode? An example is docker without a display linked into the environment.

gbalke, 2021-10-14:
@jasmeet0915 I'm going to close this issue given the length of time since my response. Please re-open if the problem persists.

TjeerdvanDijk, 2024-03-23:
the issue still persists

TjeerdvanDijk, 2024-03-23:
```
rviz2
QSocketNotifier: Can only be used with threads started with QThread
Gtk-Message: 14:08:34.886: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:08:34.887: Failed to load module "canberra-gtk-module"
[INFO] [1711199315.255069893] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1711199315.255137582] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ERROR] [1711199315.260691133] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260760838] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260800828] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260818172] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260890541] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260907247] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260943975] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260959635] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.260990328] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261005291] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261033294] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261047838] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261071201] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261086348] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261140903] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261156581] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261180762] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.261195336] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257) [... cut ...]
[ERROR] [1711199315.274499749] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1711199315.274508728] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
Aborted (core dumped)
```

Sir-Photch, 2024-04-07:
also having this issue.

hidara2000, 2024-04-25:
I was having a similar issue where I was running ROS2 in a docker container and trying to connect over SSH in Windows 11. xeyes would render but not rviz. The approach below got it working

## Dockerfile - append to end
```

RUN apt update \
    && apt install -y firefox \
                      openssh-server \
                      xauth \
    && mkdir /var/run/sshd \
    && mkdir /root/.ssh \
    && chmod 700 /root/.ssh \
    && ssh-keygen -A \
    && sed -i "s/^.*PasswordAuthentication.*$/PasswordAuthentication no/" /etc/ssh/sshd_config \
    && sed -i "s/^.*X11Forwarding.*$/X11Forwarding yes/" /etc/ssh/sshd_config \
    && sed -i "s/^.*X11UseLocalhost.*$/X11UseLocalhost no/" /etc/ssh/sshd_config \
    && grep "^X11UseLocalhost" /etc/ssh/sshd_config || echo "X11UseLocalhost no" >> /etc/ssh/sshd_config
RUN export LANG=C
RUN echo "<PUBLIC SSH KEY HERE>" >> /root/.ssh/authorized_keys

ENTRYPOINT ["sh", "-c", "/usr/sbin/sshd && tail -f /dev/null"]
```
ensuring you replace PUBLIC SSH KEY HERE with your public key for ssh

## docker-compose.yaml (snippet not complete) - ensure the following settings are included in your docker-compose file
```
... 

    build:
        context: ./
        dockerfile: Dockerfile
    
    stdin_open: true
    restart: always
    # network_mode: host
    ipc: host
    environment:
      - DISPLAY=$DISPLAY
      - XAUTHORITY=$XAUTHORITY
    
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix
      
    tty: true
    ports:
       - "2150:22"
...

## ssh config file for vscode
```
Host <your hostname>
  HostName <your hostname>
  ForwardX11 yes
  User arctos
  ForwardX11Trusted yes
  ForwardAgent yes
  XAuthLocation /opt/X11/bin/xauth
```
replace whatever your devices hostname /IP is

lastly in windows I change the default settings for Xming
<img width="800" alt="image" src="https://github.com/ros2/rviz/assets/15170494/a49b253d-fbd2-45fa-baf0-d572b9e6fcba">

Note the arrow above. I'm using 10 as a display id so I had to export this setting in the docker host
```
EXPORT DISPLAY=localhost:10.0
```

iampaulidrobo, 2024-06-09:
I think recent updates to wayland GUI doesn't support this.You have to install X11 and then proceed.
Check :
echo $XDG_SESSION_TYPE
wayland
2)Install x11-sudo apt-get install xorg openbox
3)Logout and now launch user with xorg.

Trying starting rviz2 now.

tqolf, 2025-01-07:
> I think recent updates to wayland GUI doesn't support this.You have to install X11 and then proceed. Check : `echo $XDG_SESSION_TYPE` _wayland_
> 
> 2)Install x11-sudo apt-get install xorg openbox 3)Logout and now launch user with xorg.
> 
> Try starting rviz2 now.

`echo $XDG_SESSION_TYPE` got tty after install xorg