ros2/rviz #674: Rviz2 keeps giving message filter dropping message for reason unknown

sisaha9, 2021-04-21:
[image]

I use data coming from the [SVL simulator](http://svlsimulator.com/docs/system-under-test/simulator-messages/) but when I try visualizing in Rviz I keep getting the error above. I can confirm the data is actually being published and the frames it mentions exists either via robot_state_publisher or robot_localization. Not sure what's raising the issue here

sisaha9, 2021-04-21:
Here is my TF tree for reference

[image]

Aerones-git, 2021-04-26:
I'm also having similar issue
```
[INFO] [1619443369.515854324] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443359.495 for reason 'Unknown'
[INFO] [1619443370.508120856] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443360.495 for reason 'Unknown'
[INFO] [1619443371.531937095] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443361.495 for reason 'Unknown'
[INFO] [1619443372.523825584] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443362.495 for reason 'Unknown'
[INFO] [1619443373.515947827] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443363.495 for reason 'Unknown'
[INFO] [1619443374.507506648] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443364.495 for reason 'Unknown'
[INFO] [1619443375.532200265] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443365.495 for reason 'Unknown'
[INFO] [1619443376.523829462] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443366.495 for reason 'Unknown'
[INFO] [1619443377.515699197] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443367.495 for reason 'Unknown'
[INFO] [1619443378.508079858] [rviz]: Message Filter dropping message: frame 'imu_link' at time 1619443368.495 for reason 'Unknown'

```

Using ROS2. Code:

```
        mag_msg = MagneticField()
        mag_msg.header.stamp = self.get_clock().now().to_msg()
        mag_msg.header.frame_id = self.frame_id

        mag = [float(d - 65536 if d > 32767 else d) for d in \
            [(data[0x06] << 8) + data[0x07], (data[0x08] << 8) + data[0x09], (data[0x0a] << 8) + data[0x0b]]
        ]

        mag_msg.magnetic_field.x = mag[0]
        mag_msg.magnetic_field.y = mag[1]
        mag_msg.magnetic_field.z = mag[2]

        self.mag_publisher.publish(mag_msg)
```

Hence, I believe it has nothing to do with rviz2 because the source of message can be found in https://github.com/ros2/geometry2/blob/bd125c5af32e2c63c70b7e0ba88e57094714b9b8/tf2_ros/include/tf2_ros/message_filter.h

sisaha9, 2021-05-06:
It would seem so. I can confirm rviz2 is working on another simulation

closed 2021-05-06 by sisaha9

rythm07, 2021-07-19:
@sisaha9 have you solved the problem? I'm having the same problem with SVL simulator too and it's exactly the lidar that is causing the issue

sisaha9, 2021-07-19:
In my case it was simulator lag. Is your machine able to run it frequently enough?

rythm07, 2021-07-19:
@sisaha9 I think so. My laptop has a ryzen 7 CPU and RTX2060 mobile. Isn't that enough?