ros2/rviz #676: DELETEALL Marker doesn't work for individual markers only for markerarrays

Kaju-Bubanja, 2021-04-30:
I have following two functions:

```
void delete_all_marker(rclcpp_lifecycle::LifecyclePublisher<visualization_msgs::msg::MarkerArray>::SharedPtr pub){
    pub->on_activate();
    visualization_msgs::msg::MarkerArray delete_all_marker_array;
    visualization_msgs::msg::Marker delete_all_marker;
    delete_all_marker.action = visualization_msgs::msg::Marker::DELETEALL;
    delete_all_marker_array.markers.push_back(delete_all_marker);
    pub->publish(delete_all_marker_array);
    pub->on_deactivate();
}

void delete_all_marker(rclcpp_lifecycle::LifecyclePublisher<visualization_msgs::msg::Marker>::SharedPtr pub){
    pub->on_activate();
    visualization_msgs::msg::Marker delete_all_marker;
    delete_all_marker.action = visualization_msgs::msg::Marker::DELETEALL;
    pub->publish(delete_all_marker);
    pub->on_deactivate();
}
```

and call them with one array marker pub and one marker pub

```
delete_all_marker(foo_marker_pub_);
delete_all_marker(foo_array_marker_pub_);
```

The markers from the array pub get deleted but not the one from the normal publisher. I would assume this to be a bug, but maybe I'm also using it wrong

ahcorde, 2021-06-08:
@Kaju-Bubanja 

When you try to remove the individual marker you should include the `frame_id`. Otherwise you should see something like this in the terminal:

```bash
[INFO] [1623163214.766270264] [rviz]: Message Filter dropping message: frame '' at time 0.000 for reason 'the frame id of the message is empty'
```

It's just a small addition:

```c++
void delete_all_marker(rclcpp_lifecycle::LifecyclePublisher<visualization_msgs::msg::Marker>::SharedPtr pub){
    pub->on_activate();
    visualization_msgs::msg::Marker delete_all_marker;
    delete_all_marker.header.frame_id = <your frame_id>
    delete_all_marker.action = visualization_msgs::msg::Marker::DELETEALL;
    pub->publish(delete_all_marker);
    pub->on_deactivate();
}
```

Please close the issue if this answer resolved your issue.

Kaju-Bubanja, 2021-06-08:
Thanks a bunch, this fixes the issue. I tried to add the time and thought maybe it's that, but didn't think of adding the frame id. Maybe it could be changed such that it works without? As is done for the arrays?

closed 2021-06-08 by Kaju-Bubanja