ros2/rviz #678: rviz_rendering fails to build from source on Windows 10

cardboardcode, 2021-05-01:
Hi all, 

First of all, thank you for maintaining this package. 😄

## Issue Description
Been trying to build ROS2 from source on **Windows 10**, following the instructions on [official ROS2 documentations](https://docs.ros.org/en/foxy/Installation/Windows-Development-Setup.html)

However, running `colcon build --merge-install` as instructed under the section, **[Test and Run](https://docs.ros.org/en/foxy/Installation/Windows-Development-Setup.html#test-and-run)** always **fails** at compiling **rviz_rendering** ROS2 package. 

Therefore, I tried running `colcon build --packages-select rviz_rendering --event-handlers console_direct+ --merge-install` to produce a verbose terminal output of the compilation error, as attached below.

```cmd
C:\dev\ros2_foxy>colcon build --packages-select rviz_rendering --event-handlers console_direct+ --merge-install
Starting >>> rviz_rendering
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- Found ament_cmake: 0.9.8 (C:/dev/ros2_foxy/install/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rviz_ogre_vendor: 8.2.1 (C:/dev/ros2_foxy/install/share/rviz_ogre_vendor/cmake)
-- Prepending to CMAKE_MODULE_PATH: 'C:/dev/ros2_foxy/install/share/rviz_ogre_vendor/cmake/../../../opt/rviz_ogre_vendor/CMake'
-- Setting OGRE_DIR to: 'C:/dev/ros2_foxy/install/share/rviz_ogre_vendor/cmake/../../../opt/rviz_ogre_vendor/CMake'
-- Found OGRE
--   static     : OFF
--   components : HLMS;MeshLodGenerator;Overlay;Paging;Property;RTShaderSystem;Terrain;Volume
--   plugins    : Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus;Codec_STBI
--   media      : C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/Media
-- OGRE_LIBRARIES: OgreHLMS;OgreMeshLodGenerator;OgreOverlay;OgrePaging;OgreProperty;OgreRTShaderSystem;OgreTerrain;OgreVolume;OgreMain
-- OGRE_LIBRARY_DIRS: C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/lib
-- OGRE_PLUGINS: Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus;Codec_STBI
-- OGRE_PLUGIN_DIR: C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/lib/OGRE
-- rviz_ogre_vendor::OgreOverlay for IMPORTED_LOCATION_RELEASE: C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/lib/OgreOverlay.lib
-- rviz_ogre_vendor::OgreOverlay for IMPORTED_LOCATION_DEBUG: _ogre_overlay_static_library_debug_abs-NOTFOUND
-- rviz_ogre_vendor::OgreMain for IMPORTED_LOCATION_RELEASE: C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/lib/OgreMain.lib
-- rviz_ogre_vendor::OgreMain for IMPORTED_LOCATION_DEBUG: _ogre_main_static_library_debug_abs-NOTFOUND
-- Could not find freeimage library
-- rviz_ogre_vendor::RenderSystem_GL for IMPORTED_LOCATION_RELEASE: C:/dev/ros2_foxy/install/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.lib
-- rviz_ogre_vendor::RenderSystem_GL for IMPORTED_LOCATION_DEBUG: _render_system_gl_static_library_debug_abs-NOTFOUND
-- Found rviz_assimp_vendor: 8.2.1 (C:/dev/ros2_foxy/install/share/rviz_assimp_vendor/cmake)
-- library: assimp-vc140-mt
-- Found ament_index_cpp: 1.0.2 (C:/dev/ros2_foxy/install/share/ament_index_cpp/cmake)
-- Found eigen3_cmake_module: 0.1.1 (C:/dev/ros2_foxy/install/share/eigen3_cmake_module/cmake)
-- Found resource_retriever: 2.3.4 (C:/dev/ros2_foxy/install/share/resource_retriever/cmake)
-- library: C:/dev/ros2_foxy/install/opt/libcurl_vendor/bin/libcurl.dll
-- Found ament_cmake_cppcheck: 0.9.6 (C:/dev/ros2_foxy/install/share/ament_cmake_cppcheck/cmake)
-- Found ament_cmake_cpplint: 0.9.6 (C:/dev/ros2_foxy/install/share/ament_cmake_cpplint/cmake)
-- Found ament_cmake_lint_cmake: 0.9.6 (C:/dev/ros2_foxy/install/share/ament_cmake_lint_cmake/cmake)
-- Found ament_cmake_uncrustify: 0.9.6 (C:/dev/ros2_foxy/install/share/ament_cmake_uncrustify/cmake)
-- Found ament_cmake_gmock: 0.9.8 (C:/dev/ros2_foxy/install/share/ament_cmake_gmock/cmake)
-- Found ament_cmake_gtest: 0.9.8 (C:/dev/ros2_foxy/install/share/ament_cmake_gtest/cmake)
-- Found gmock sources under 'C:/dev/ros2_foxy/install/src/gmock_vendor': C++ tests using 'Google Mock' will be built
-- Found gtest sources under 'C:/dev/ros2_foxy/install/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/ros2_foxy/build/rviz_rendering
Microsoft (R) Build Engine version 16.9.0+5e4b48a27 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  gmock.vcxproj -> C:\dev\ros2_foxy\build\rviz_rendering\gmock\Release\gmock.lib
  gmock_main.vcxproj -> C:\dev\ros2_foxy\build\rviz_rendering\gmock\Release\gmock_main.lib
  Building Custom Rule C:/dev/ros2_foxy/src/ros2/rviz/rviz_rendering/CMakeLists.txt
     Creating library C:/dev/ros2_foxy/build/rviz_rendering/Release/rviz_rendering.lib and object C:/dev/ros2_foxy/build/rviz_rendering/Release/rviz_rendering.exp
moc_render_window.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: static struct QMetaObject const QWindow::staticMetaObject" (__imp_?staticMetaObject@QWindow@@2UQMetaObject@@B) referenced in function "void __cdecl `dynamic initializer for 'public: static struct QMetaObject const rviz_rendering::RenderWindow::staticMetaObject''(void)" (??__E?staticMetaObject@RenderWindow@rviz_rendering@@2UQMetaObject@@B@@YAXXZ) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
moc_render_window.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: class QString __cdecl QMetaObject::tr(char const *,char const *,int)const " (__imp_?tr@QMetaObject@@QEBA?AVQString@@PEBD0H@Z) referenced in function "public: static class QString __cdecl rviz_rendering::RenderWindow::tr(char const *,char const *,int)" (?tr@RenderWindow@rviz_rendering@@SA?AVQString@@PEBD0H@Z) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
viewport_projection_finder.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: class QString __cdecl QMetaObject::tr(char const *,char const *,int)const " (__imp_?tr@QMetaObject@@QEBA?AVQString@@PEBD0H@Z) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
render_window.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: class QString __cdecl QMetaObject::tr(char const *,char const *,int)const " (__imp_?tr@QMetaObject@@QEBA?AVQString@@PEBD0H@Z) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
moc_render_window.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: struct QMetaObject * __cdecl QObjectData::dynamicMetaObject(void)const " (__imp_?dynamicMetaObject@QObjectData@@QEBAPEAUQMetaObject@@XZ) referenced in function "public: virtual struct QMetaObject const * __cdecl rviz_rendering::RenderWindow::metaObject(void)const " (?metaObject@RenderWindow@rviz_rendering@@UEBAPEBUQMetaObject@@XZ) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
moc_render_window.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual void * __cdecl QWindow::qt_metacast(char const *)" (__imp_?qt_metacast@QWindow@@UEAAPEAXPEBD@Z) referenced in function "public: virtual void * __cdecl rviz_rendering::RenderWindow::qt_metacast(char const *)" (?qt_metacast@RenderWindow@rviz_rendering@@UEAAPEAXPEBD@Z) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
moc_render_window.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual int __cdecl QWindow::qt_metacall(enum QMetaObject::Call,int,void * *)" (__imp_?qt_metacall@QWindow@@UEAAHW4Call@QMetaObject@@HPEAPEAX@Z) referenced in function "public: virtual int __cdecl rviz_rendering::RenderWindow::qt_metacall(enum QMetaObject::Call,int,void * *)" (?qt_metacall@RenderWindow@rviz_rendering@@UEAAHW4Call@QMetaObject@@HPEAPEAX@Z) [C:\dev\ros2_foxy\build\rviz_rendering\rviz_rendering.vcxproj]
mesh_loader.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl QString::~QString(void)" (__imp_??1QString@@QEAA@XZ) referenced in function "class Ogre::SharedPtr<class Ogre::Mesh> __cdecl rviz_rendering::loadMeshFromResource(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?loadMeshFromResource@rviz_rendering@@YA?AV?$SharedPtr@VMesh@Ogre@@@Ogre@@AEBV?$bas [... cut ...]
## Tried Solution

I have looked at similar past issues that have been resolved before and tried the following under [Issue 543](https://github.com/ros2/ros2/issues/543) of `ros2` repository.
> 
    Using this stripped down cygwin version of patch: patch-2.7.4-cygwin.zip
    Using this updated patch file in rviz_ogre_vendor: ogre-ca665a6-patch-fix.diff.txt

However, none seems to work and I am stucked. 😞 

## Others
Appreciate any help with this issue. Do let me know if more details are required.

flylinux, 2021-09-21:
set Qt5_DIR=C:\Qt\5.15.2\msvc2019_64

clalancette, 2022-02-03:
Closing this based on the last comment.

closed 2022-02-03 by clalancette