ros2/rviz #680: Path msg orientation error

BriceRenaudeau, 2021-05-05:
When displaying a `nav_msgs::msg::Path` with `frame_id = "odom"` using "map" as the fixed frame, the orientation of the poses is wrong.
The display depends on the orientation of the "map" frame.

[image]

[image]

When displaying it in the "odom" frame, they are in the good orientation.

[image]

rebecca-butler, 2021-07-28:
@BriceRenaudeau could you please provide a code sample to reproduce this issue? Thanks!

BriceRenaudeau, 2021-07-28:
This simple node publishes a path every 5s.
```c++
class MinimalPub: public rclcpp::Node
{
  public:
    // constructor
    MinimalPub()
    : Node("minimal_publisher"), count(0)
    {
      publisher = this->create_publisher<nav_msgs::msg::Path>("path", rclcpp::QoS(1).transient_local()); // topic latched
      timer_callback();
      timer = this->create_wall_timer(5s, std::bind(&MinimalPub::timer_callback, this));
    }

    //methodes
    void timer_callback()
    {
      auto message = nav_msgs::msg::Path();
      message.header.stamp = now();
      message.header.frame_id = "odom";
      geometry_msgs::msg::PoseStamped pose;
      pose.header.frame_id = "odom";
      for (unsigned int i=0; i<10; i++) {
        pose.pose.position.x += 1.0;
        message.poses.push_back(pose);
      }
      publisher->publish(message);
    }

    // attributes
    rclcpp::TimerBase::SharedPtr timer;
    rclcpp::Publisher<nav_msgs::msg::Path>::SharedPtr publisher;
    size_t count;
};
```

Then you can try publishing /TF from map to odom with different orientations.

PR #736 (2021-08-04): Fix path message orientation error
Closes #680 

This PR fixes the orientation of the arrow and axes markers for path messages.
closed 2021-08-04 by rebecca-butler