ros2/rviz #693: rviz2 crash when subscribe to path msg remotely

hoppss, 2021-05-21:
rviz2 crash when i add a Path plugin which subscribe to "/local_plan" msg that is published by local planner.
error msg is shown as following

```
[ERROR] [1621564700.523254749] [rviz2]: InternalErrorException: Cannot create GL vertex buffer in GLHardwareVertexBuffer::GLHardwareVertexBuffer at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.1/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GL/src/OgreGLHardwareVertexBuffer.cpp (line 46)
terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  InternalErrorException: Cannot create GL vertex buffer in GLHardwareVertexBuffer::GLHardwareVertexBuffer at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.1/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GL/src/OgreGLHardwareVertexBuffer.cpp (line 46)
Aborted (core dumped)
```
os: ubuntu20.04
ros: foxy
navigation2 is runing at a real robot and rviz2 is running at my laptop, communicate by same wifi and same ROS_DOMAIN_ID;
rviz2 must crash when i subsribe to local_plan topic, but it is ok when i run tb3 gazebo simulation, its  strange...

[image]

hoppss, 2021-06-03:
it might be caused by different frame trasformation calculation with tf and different stamp between robot onborad pc and remote laptop system clock

padhupradheep, 2021-07-08:
I'm also facing the same issue, is there any update ?

Serafadam, 2021-07-09:
Same issue here, I'm wondering if this is related to https://github.com/ros2/rviz/issues/691

PSotiriadis, 2021-07-27:
I am facing exactly the same problem. RVIZ2 crashes, with the same error, if I visualize /local_plan topic. If not, everything is OK and I can navigate, giving waypoints.
Additionally, before this error, I also receive the following error, relative to GL:
 
```
[rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
active samplers with a different type refer to the same texture image unit
```
I don't know if both errors are connected...

I have also Ubuntu 20.04 with foxy. I start also SLAM & Navigation on real robot and on my desktop PC I start only RVIZ2.
If I start also RVIZ2 on real robot, then everything is working properly, also /local_plan topic visualization.

Serafadam, 2021-07-28:
@PSotiriadis I stopped having this issue after reinstalling Linux & switching to Galactic, maybe this will help

PSotiriadis, 2021-07-30:
Hi Serafadam, First thank you very much for your fast reply. Jo, thats a good tipp and it really helps me. It is in my plans to jump to galactic but now, I see, I need to do it faster 😀Have a nice day and a nice weekend!!!Sent from my Galaxy

padhupradheep, 2021-08-30:
Problem still persists in Foxy, in galactic and rolling not!

gbalke, 2021-09-24:
If someone could provide a quick demo of the bug, I have some time to dive in and find a fix. It'll take some more time to create a demo for myself.

clalancette, 2023-07-14:
Given that Foxy is now End-of-Life, and the bug seems to be fixed in later versions, I'm going to close this out.  If you are still seeing this in Humble or Iron, please feel free to open a new issue.

closed 2023-07-14 by clalancette