ros2/rviz #699: Break fork from ros-visualization/rviz if not being reintegrated

SteveMacenski, 2021-06-03:
Hi,

I wanted to request a fork break (like we did with geometry2 https://github.com/ros2/geometry2/issues/237) so that we can use the GitHub search functionality to find function signatures and quickly track down implementations. With Forks in GitHub, this functionality is disabled and only searches the parent repository. Given that there are some changes, it would be nice if that functionality was restored.

If you email the GitHub folks, they'll sever a fork for you into its same repo, same name, and allows this feature to come back. I did this with slam toolbox and geometry2 which originally started as a fork of slam karto.

https://support.github.com/contact

clalancette, 2021-06-17:
I've now requested GitHub to break the fork.

clalancette, 2021-06-18:
All right, this has now been done.

closed 2021-06-18 by clalancette