ros2/rviz #703: rviz2 randomly crashes while running with nav2 stack

AlexeyMerzlyakov, 2021-06-08:
### Basic info
RViz2 from frequently crashes during the use while running with Nav2 stack. The problem appears in random places even if all Nav2-related stuff was disabled (see minimal configuration file in reproduction section below). Crashes were [noted](https://github.com/ros-planning/navigation2/issues/2367) on Crystal and still appearing on latest Rolling.

### More info
Crashes appears irregularly in different places. All types of messages and backtraces were detected and sorted-out by their probability of occurrence:
1. Invalid Argument
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
```
2. Segfault
```
Thread 16 "rviz2" received signal SIGSEGV, Segmentation fault.
```
3. Assertion Robust
```
rviz2: ../nptl/pthread_mutex_lock.c:428: __pthread_mutex_lock_full: Assertion `e != ESRCH || !robust' failed.
--Type <RET> for more, q to quit, c to continue without paging--
```
4. Deadlock: RViz2 hangs during the use
5. Assertion Prio
```
[rviz2-4] rviz2: tpp.c:82: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
[ERROR] [rviz2-4]: process has died [pid 174985, exit code -6, cmd '/home/leha/ros2_rolling/install/rviz2/lib/rviz2/rviz2 -d /home/leha/nav2_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args'].
```

For the types No.1-4 GDB backtraces were collected and attached in the archive: [backtraces.zip](https://github.com/ros2/rviz/files/6618073/backtraces.zip)
For type No.5 backtrace was not obtained (it was detected once). All of the problems above were detected by using the same scenario. Thus maybe (but not proven) all issue types have the same root cause.

### Steps to reproduce

Despite the problem is irregular and there is no 100% algorithm how to reproduce it, here are the steps allowing with more or less probability to get the problem:
1. Replace `navigation2/nav2_bringup/bringup/rviz/nav2_default_view.rviz` file with minimal repro-configuraiton attached to current issue: [nav2_default_view.rviz.zip](https://github.com/ros2/rviz/files/6618076/nav2_default_view.rviz.zip)
2. Run Nav2 simulation as usual:
```
ros2 launch nav2_bringup tb3_simulation_launch.py
```
3. In appeared RViz screen set robot pose by "2D Pose Estimate"
4. enable-disable many times the "Controller" checkbox as shown in the picture below:
[image]
5. If problem won't appear within ~30-40 seconds, close RViz and repeat starting from step No.2.

### Environment
* Ubuntu 20.04 x64
* ROS2 Distro: Rolling built from sources (`ros2.repos` from Jun 1st).
* Nav2 main and Nav2 dependencies were [built from sources](https://navigation.ros.org/build_instructions/index.html#manually-build-main). Nav2 commit: https://github.com/ros-planning/navigation2/commit/b63b11f8a638faf389d6b99911d95899abe1580b
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
RViz commit: 94f6dd6958b2927a307893ff174f182dcc7afd0e
```
$ ros2 run rviz2 rviz2 -l
...
[INFO] [1623165832.507667729] [rviz2]: *-*-* OGRE Initialising
[INFO] [1623165832.507680536] [rviz2]: *-*-* Version 1.12.1 (Rhagorthua)
...
[INFO] [1623166434.886964038] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
```

Qt version is not printed by rviz:
```
$ qmake --version
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
```
* If source build, which git commit?
The problem is long-standing detected since Crystal. Accused commit was not found.

clalancette, 2021-06-25:
So I spent a bit of time looking at this yesterday.

First, thanks to @AlexeyMerzlyakov for providing easy instructions for a reproducer.  Following the instructions in the initial comment, I was able to reproduce after a few tries.

I then rebuilt parts of the stack with debugging enabled.  After adding in some debugging here and there, my initial thought is that this is actually a bug in `tf2_ros` that is getting exposed by repeatedly enabling and disabling the Map and/or the Polygon in the UI.

In particular, it seems like what is happening is that every time the Map or Polygon is getting enabled, it is creating a new `MessageFilter` object.  Every time they are disabled, that `MessageFilter` object is destroyed.  One of the things that can happen while the `MessageFilter` object is alive is that it can start waiting for a transform: https://github.com/ros2/geometry2/blob/c36999b17cbf1e64a7937232133db79898bfa608/tf2_ros/include/tf2_ros/message_filter.h#L426 .  However, the kicker is that if the `MessageFilter` object is destroyed before that transform comes in, it never unregisters its interest in that transform.  Thus, the `MessageFilter` object may be destroyed, and sometime later the underlying `Buffer` object may get the transform and call `transformReadyCallback` for that transform.  But that is now pointing to random memory, so arbitrary things may happen.

I played around a bit with trying to unregister the wait when the `MessageFilter` object was destroyed, but I wasn't really successful yet.  I'm going to have to put this down for now, but I think this is the direction we need to explore; make sure that the `MessageFilter` unregisters any `waitForTransforms` it did while it was active.

iuhilnehc-ynos, 2021-12-22:
> However, the kicker is that if the `MessageFilter` object is destroyed before that transform comes in, it never unregisters its interest in that transform. Thus, the `MessageFilter` object may be [...]

I agree with what @clalancette analyzed about the root cause for the issue.
I'd like to share something about my thought. If MessageFilter is destroyed, I think we can do 
- to clear all relative callbacks in the `transformable_callbacks_` of BufferCore. (easy to cause a deadlock inside.)
- to use a mechanism to call the callback `transformReadyCallback` but ignore doing the real thing that accessing the members of the destroyed instance.

<details><summary> patches for some repositories </summary>
<p>

- https://github.com/ros2/geometry2.git

```patch
diff --git a/tf2_ros/include/tf2_ros/message_filter.h b/tf2_ros/include/tf2_ros/message_filter.h
index cb61843..87c90e7 100644
--- a/tf2_ros/include/tf2_ros/message_filter.h
+++ b/tf2_ros/include/tf2_ros/message_filter.h
@@ -144,7 +144,8 @@ public:
  \endverbatim
  */
 template<class M, class BufferT = tf2_ros::Buffer>
-class MessageFilter : public MessageFilterBase, public message_filters::SimpleFilter<M>
+class MessageFilter : public MessageFilterBase, public message_filters::SimpleFilter<M>,
+                      public std::enable_shared_from_this<MessageFilter<M, BufferT>>
 {
 public:
   using MConstPtr = std::shared_ptr<M const>;
@@ -428,7 +429,10 @@ public:
         frame_id,
         stamp,
         buffer_timeout_,
-        std::bind(&MessageFilter::transformReadyCallback, this, std::placeholders::_1, handle));
+        std::bind(&MessageFilter::transformReadyCallback,
+          this, std::placeholders::_1, handle,
+          // use this->weak_from_this() if C++17 is enable
+          std::weak_ptr<MessageFilter<M, BufferT>>(this->shared_from_this())));
     }
   }
 
@@ -478,95 +482,101 @@ private:
     expected_success_count_ = 1;
   }
 
-  void transformReadyCallback(const tf2_ros::TransformStampedFuture & future, const uint64_t handle)
+  void transformReadyCallback(
+    const tf2_ros::TransformStampedFuture & future,
+    const uint64_t handle,
+    std::weak_ptr<MessageFilter<M, BufferT>> weak_this)
   {
     namespace mt = message_filters::message_traits;
 
     MEvent saved_event;
     bool event_found = false;
 
-    {
-      // We will be accessing and mutating messages now, require unique lock
-      std::unique_lock<std::mutex> lock(messages_mutex_);
-
-      // find the message this request is associated with
-      typename L_MessageInfo::iterator msg_it = messages_.begin();
-      typename L_MessageInfo::iterator msg_end = messages_.end();
-
-      for (; msg_it != msg_end; ++msg_it) {
-        MessageInfo & info = *msg_it;
-        auto handle_it = std::find(info.handles.begin(), info.handles.end(), handle);
-        if (handle_it != info.handles.end()) {
-          // found msg_it
-          ++info.success_count;
-          if (info.success_count >= expected_success_count_) {
-            saved_event = msg_it->event;
[... 184 lines cut ...]
-  std::unique_ptr<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>> scan_filter_;
+  std::shared_ptr<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>> scan_filter_;
   std::shared_ptr<rclcpp::Publisher<nav_msgs::msg::OccupancyGrid>> sst_;
   std::shared_ptr<rclcpp::Publisher<nav_msgs::msg::MapMetaData>> sstm_;
   std::shared_ptr<rclcpp::Service<nav_msgs::srv::GetMap>> ssMap_;
diff --git a/src/slam_toolbox_common.cpp b/src/slam_toolbox_common.cpp
index 8d01022..a27bec0 100644
--- a/src/slam_toolbox_common.cpp
+++ b/src/slam_toolbox_common.cpp
@@ -208,7 +208,7 @@ void SlamToolbox::setROSInterfaces()
     std::make_unique<message_filters::Subscriber<sensor_msgs::msg::LaserScan>>(
     shared_from_this().get(), scan_topic_, rmw_qos_profile_sensor_data);
   scan_filter_ =
-    std::make_unique<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>>(
+    std::make_shared<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>>(
     *scan_filter_sub_, *tf_, odom_frame_, 1, shared_from_this(),
     tf2::durationFromSec(transform_timeout_.seconds()));
   scan_filter_->registerCallback(
```

</p>
</details>

This patch does not extend the lifecycle of `MessageFilter`.
Could somebody help confirm it?

clalancette, 2021-12-22:
@iuhilnehc-ynos Thanks so much for taking a look at this!

To test your code, I first recompiled all of the ROS 2 core from source, and navigation2 from source.  With that configuration (and the reproducer provided in the first comment), I was able to reproduce the error.

I then applied your code to geometry2 and to navigation2 (I wasn't using slam_toolbox, so I didn't test that code).  I then recompiled both the ROS 2 core and navigation2.  With that code in place, I was not able to reproduce the crash after trying 10 times locally.  This doesn't guarantee that it is fixed, but it is a really good indication.

@iuhilnehc-ynos Do you mind opening PRs for these?

iuhilnehc-ynos, 2021-12-23:
@clalancette 

> With that code in place, I was not able to reproduce the crash after trying 10 times locally. This doesn't guarantee that it is fixed, but it is a really good indication.

Thank you for your confirmation. Before opening PRs, I need to try to figure out the first method which seems ideal.
(The `geometry2` implementation in [ROS2](https://github.com/ros2/geometry2.git) is a little different from [ROS1](https://github.com/ros/geometry2.git))

iuhilnehc-ynos, 2021-12-23:
I have done the first method. Even if it seems to work well, I need to check it more.
The basic idea is as follows,

- update `TransformStampedFuture` 

NOTE: not to use `std::pair<TransformStampedFuture, tf2::TransformableRequestHandle>`

from
```cpp
using TransformStampedFuture = std::shared_future<geometry_msgs::msg::TransformStamped>;
```
to
```cpp
class TransformStampedFuture : public std::shared_future<geometry_msgs::msg::TransformStamped> {
 ...
 private:
  tf2::TransformableRequestHandle handle_;
}
```

- add a new interface in `AsyncBufferInterface`

```cpp
  virtual void
  cancel(TransformStampedFuture& ts_future) = 0;
```

iuhilnehc-ynos, 2021-12-23:
@clalancette 

I am not familiar with both geometry2 and rviz2. I don't know if I am on the correct way to fix this issue.
Could you help review these two PRs above?

closed 2022-02-17 by clalancette