ros2/rviz #704: Attaching a 4th monitor lags RVIZ2 with nouveau drivers

Kaju-Bubanja, 2021-06-09:
I am running the master branch of ROS2 and RVIZ2. Everything works fine until I plug in my 4th monitor, then RVIZ starts to lag really hard, having around 0.2 fps. I confirmed that it is the monitor and the nouveau driver, because after unplugging the 4th monitor everything works fine. I now switched to the nvidia drivers and again everything works fine, also with 4 monitors.

System info: Ubuntu 20.04
I also opened a nouveau issue, for reference: https://gitlab.freedesktop.org/nouveau/wiki/-/issues/5

clalancette, 2021-06-09:
Given that this looks to be a problem in the drivers (and not in RViz2 itself), I'm going to go ahead and close this.  Feel free to reopen if you think this is a problem in RViz2.

closed 2021-06-09 by clalancette

Kaju-Bubanja, 2021-06-10:
I'm not sure it's only in the drivers. Because other windows don't lag, like firefox, pycharm, teams. It's just rviz. Even though the issue is clearly somewhere in the driver, I think it has to do how rviz uses them. Also I don't have the rights to open the issue once closed

clalancette, 2021-06-10:
> I'm not sure it's only in the drivers. Because other windows don't lag, like firefox, pycharm, teams. It's just rviz. Even though the issue is clearly somewhere in the driver, I think it has to do with how rviz uses them.

One large difference between those other things and RViz is probably that RViz uses OGRE for the rendering.  It might be interesting to test another project using OGRE (like Gazebo or Ignition), and see if they similarly lag.

> Also I don't have the rights to open the issue once closed

Huh, I didn't realize that was the case.  I have to say, that policy decision from GitHub doesn't make a whole lot of sense to me; the reporter could just open another issue :).  Anyway, I'm still going to leave this closed for now, and if we think there is really something to fix in RViz2 itself, I'll go ahead and reopen.