ros2/rviz #712: support ascii stl files

BrettRD, 2021-06-21:
I'm using rviz installed from ubuntu 20.04 binaries at foxy, trying to load a robot model exported from openscad.

I'm getting an error with:
`... rviz can only load binary STL files so it will not be loaded. Please convert it to a binary STL file.`
similar to rviz https://github.com/ros-visualization/rviz/issues/1364
which apparently is fixed in https://github.com/ros-visualization/rviz/pull/1365

clalancette, 2021-06-23:
If you'd like to open a PR to port that functionality to rviz2, that would be most welcome.  Otherwise, we'll keep it on the TODO list, but we have no guarantees of when it will get done.

adamkrawczyk, 2021-09-01:
Hi, 
I would like to take a look here. Where should I start? 

Best

BrettRD, 2021-09-03:
Anything exported from openscad should trip up rviz2
These two repos (one is a submodule of the other) will spool a `robot_description` for rviz2 to parse, (I've forgotten how I converted to binary previously)
https://github.com/BrettRD/dragon_claw
https://github.com/BrettRD/kinematics_scad

I didn't get around to trying this, but I was planning on seeing if 
https://github.com/ros-visualization/rviz/pull/1365 would cherry-pick cleanly

ahcorde, 2023-09-11:
Related PR https://github.com/ros2/rviz/pull/1063

closed 2023-10-16 by ahcorde