ros2/rviz #720: Rviz2 for the browser

Kaju-Bubanja, 2021-06-28:
There are some cool projects that ported parts of rviz to the browser, unfortunately they are heavily tied to the ros binaries and not easy to port for rviz2. Then there is following package: https://github.com/emersonknapp/omni-rviz2 but it's stale. So I would like to make a feature request for rviz2 in the browser. I was wondering if this was discussed before and if somebody has an estimate of how long this would take and some high level milestones.

wolfv, 2021-07-03:
You could also have a look at the Zethus project which reimplements RViz in JS + ThreeJS (github.com/robostack/zethus)

jacobperron, 2021-07-12:
I'm not aware of any prior discussions about supporting RViz in the browser. It sounds like a large endeavor. If someone would like to make an effort towards browser support, I suggest they start with a design document (e.g. a new article [here](https://github.com/ros2/rviz/tree/ros2/docs)), describing the planned approach and what would have to change architecturally. Having a design doc reviewed and approved would make subsequent pull requests more likely to be accepted.

XDGFX, 2021-08-11:
This is something that would be useful for me - the device I'm developing on is headless and so I'm running RViz2 on another computer on the network.

This works with ROS Docker containers with host networking, but sometimes bridge networking is preferred/required, and as such a different computer on the network cannot see ROS topics inside the bridge network.

A possible solution for my use case is running a desktop environment with RViz2 inside a Docker container, and using a VNC server to connect remotely. I've tried a few examples I've found online as well as building a container myself, but haven't been able to get anything working.

clalancette, 2021-08-11:
For what it is worth, there are projects like https://webviz.io/ and https://foxglove.dev/ that do exactly this.

Whether it makes sense to port this particular RViz2 code to the browser, or just use one of those projects, isn't totally clear to me.

XDGFX, 2021-08-11:
I had seen some of these before and they look like they would work perfectly. I've been having lots of trouble getting https://github.com/RobotWebTools/rosbridge_suite/ to work, which both depend on for live viewing. Apparently an official release is happening Soon™ (https://github.com/RobotWebTools/rosbridge_suite/issues/536), so maybe when that's out I will get it working! Thanks.

jhurliman, 2021-08-11:
Foxglove developer here. If you use the desktop version, you can connect to a native ROS 1 robot without the rosbridge WebSocket bridge. Support for native connections to ROS 2 robots from the desktop app should land near the end of this month. The web version is unable to open raw TCP/UDP sockets so it relies on rosbridge for connectivity, but this will be true of any web app.

We have patched rosbridge_suite to work with ROS 2 and are in the process of releasing it for Foxy and Galactic right now. You can follow https://github.com/foxglove/studio/issues/716 for updates on this.

XDGFX, 2021-08-12:
Hi John! After a few hours of fiddling I did manage to get rosbridge_suite working with ROS2. Foxglove seems to be working for the most part, I need to do some more testing before I can confirm it'll work for my use case.

Slightly off topic, but for some reason I was getting formatting errors in the launch file, and in the Python script the `ParameterDescriptor` didn't like the `dynamic_typing` argument. The below fixed that issue for me, not sure if this is known.

```Dockerfile
RUN sed -i "s/'/''/g" src/rosbridge_suite/rosbridge_server/launch/rosbridge_websocket_launch.xml && \
    sed -i "s/ParameterDescriptor(dynamic_typing=True)/ParameterDescriptor()/g" src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket.py
```

What do you mean by native connections? The topic you linked maybe indicates Foxglove Studio will be able to ssh into e.g. a container and attach to topics as needed? That would be very nice.

jtbandes, 2021-08-12:
The ParameterDescriptor issue is known and seems to be a difference between Foxy and Galactic (assuming you’re using Foxy?) — I’ll try to get this fixed before making a Foxy release. Might ask for your help in testing it :)

XDGFX, 2021-08-12:
Ah sure, yep using Foxy. Sure, I can do my best to help :)

Gotcha. I think again that would run into the limitation that Studio would need to be on the same network - which isn't the case if using Docker bridge for containers (but can connect to services on specific ports, e.g. start an ssh server).

clalancette, 2023-07-14:
Given that there are other solutions for the original request here, I'm going to close this out.

closed 2023-07-14 by clalancette