ros2/rviz #730: Rviz2 crashes when the sparse map is empty

carlospzlz, 2021-07-20:
We are doing some live tests with our robot platform and we display the topics using `rviz2`. In one of our tests we cover the camera of the robot, therefore the sparse map that is published contains no landmarks.
`rviz2` crashes with the following error:

```
[INFO] [1626779830.868988961] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1626779830.869022514] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1626779830.884537259] [rviz2]: Stereo is NOT SUPPORTED
/usr/include/c++/11.1.0/bits/stl_vector.h:1134: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::front() const [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; std::vector<_Tp, _Alloc>::const_reference = const unsigned char&]: Assertion '!this->empty()' failed.
fish: Job 1, 'rviz2' terminated by signal SIGABRT (Abort)
```

jacobperron, 2021-07-22:
@carlospzlz Could you provide precise steps for reproducing the crash. This will significantly improve the chances of the bug being fixed. Thanks!

carlospzlz, 2021-11-08:
Hi @jacobperron, I can't reproduce this issue anymore, feel free to close it :)

closed 2021-11-08 by clalancette