ros2/rviz #737: Dashing - no camera info for camera display

Marenix, 2021-08-03:
I noticed some time ago that on Dashing camera display didn't work. I also saw that rviz was looking for camera_info on the wrong topic. I didn't go further into it, until today, and I found [PR#516](https://github.com/ros2/rviz/pull/516) which seemed to address it. Still, camera display wasn't on the newest release. Looking through the changes in that PR, I saw this:
[image]
It seems that the right camera topic name was added in variable camera_info_topic, but for some reason, the old camera_info topic name is used when creating a subscriber. Was this just mistakenly overlooked?

clalancette, 2021-08-05:
This seems to have all been fixed in later versions; see https://github.com/ros2/rviz/blob/24f617cb5e76a75ee9ed2a37ab1284ad4c686471/rviz_default_plugins/src/rviz_default_plugins/displays/camera/camera_display.cpp#L282-L303 , for instance.

Since Dashing is now EOL, we won't be doing any further updates to it.  But if you are still having problems with the supported versions (Foxy, Galactic, or Rolling), please feel free to open up another issue.

I'm going to close this out for now.

closed 2021-08-05 by clalancette

Marenix, 2021-08-06:
Yes, it has been fixed in later version. This works in every version after Dashing. I was just confused that I found a PR that "fixed" of camera display not working in Dashing, and the problem still existed in the newest Dashing release.

Anyway, thank for the help!