ros2/rviz #739: Can't link with assimp

Kaju-Bubanja, 2021-08-09:
When trying to compile rviz_rendering I get following stacktrace:
```

--- stderr: rviz_rendering
/usr/bin/ld: cannot find -lassimp
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rviz_rendering.dir/build.make:515: librviz_rendering.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:344: CMakeFiles/rviz_rendering.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
```

I have both assimp-utils and assimp-testmodels installed. This is my output for `ls -la | grep assimp`

```
lrwxrwxrwx   1 root root        18 Feb 13  2020 libassimp.so.5 -> libassimp.so.5.0.0
-rw-r--r--   1 root root  12846808 Feb 13  2020 libassimp.so.5.0.0
```

How can I get rviz_rendering to build?

clalancette, 2021-08-09:
Please open questions like this on https://answers.ros.org.  That is our central, searchable site for questions and answers.  When you open your question, please be sure to put a lot of detail in it, including what operating system you are using, what ROS version you are using, and the exact commands you are using that are failing.

I'm going to close this one out.

closed 2021-08-09 by clalancette

Kaju-Bubanja, 2021-08-10:
Deleting the rviz_rendering folder in the build folder and rebuilding the package fixed this error