ros2/rviz #741: rviz_common package failed to build: MacOS Ros2 Galactic Installation

ParthGaneriwala, 2021-08-11:
## Bug report
```
In file included from /Users/assistlab/ros2_galactic/build/rviz_common/src/rviz_common/moc_displays_panel.cpp:10:
In file included from /Users/assistlab/ros2_galactic/build/rviz_common/src/rviz_common/../../../../src/ros2/rviz/rviz_common/src/rviz_common/displays_panel.hpp:40:
In file included from /Users/assistlab/ros2_galactic/src/ros2/rviz/rviz_common/include/rviz_common/config.hpp:41:
In file included from /usr/local/opt/qt@5/lib/QtCore.framework/Headers/QVariant:1:
/usr/local/opt/qt@5/lib/QtCore.framework/Headers/qvariant.h:77:7: error: definition of type 'QStringList' conflicts with type alias of the same name
class QStringList;
      ^
/usr/local/include/QtCore/qcontainerfwd.h:65:7: note: 'QStringList' declared here
...
make[2]: *** [CMakeFiles/rviz_common.dir/src/rviz_common/moc_frame_manager.cpp.o] Error 1
11 errors generated.
make[2]: *** [CMakeFiles/rviz_common.dir/include/rviz_common/moc_frame_manager_iface.cpp.o] Error 1
8 errors generated.
make[2]: *** [CMakeFiles/rviz_common.dir/include/rviz_common/moc_visualizer_app.cpp.o] Error 1
make[1]: *** [CMakeFiles/rviz_common.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_common [10.4s, exited with code 2]
Aborted  <<< ros2run [0.90s]
Aborted  <<< rosbag2_tests [1.24s]                                           
Aborted  <<< ros2launch [0.95s]
Aborted  <<< ros2param [1.01s]                                               
Aborted  <<< ros2topic [6.98s]
Aborted  <<< sros2_cmake [1.01s]                          
Aborted  <<< ros2lifecycle [1.10s]
Aborted  <<< test_security [1.12s]                        
Aborted  <<< test_launch_ros [2.29s]                               

Summary: 286 packages finished [1min 28s]
  1 package failed: rviz_common
  9 packages aborted: ros2launch ros2lifecycle ros2param ros2run ros2topic rosbag2_tests sros2_cmake test_launch_ros test_security
  18 packages had stderr output: class_loader fastcdr orocos_kdl osrf_testing_tools_cpp performance_test_fixture rcl_logging_interface rcl_logging_log4cxx rcl_logging_noop rcl_logging_spdlog rcpputils resource_retriever rosbag2_storage rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_cpp rosidl_typesupport_fastrtps_c rviz_common urdfdom
  8 packages not processed
```
**Required Info:**

- Operating System:
  - macOS Big Sur v11.5.1
  - Processor - 3.8 GHz 8-Core Intel Core i7
  - Memory - 32 GB 2667 MHz DDR4 
- Installation type:
  - from source
- DDS implementation:
  - OpenSSL

#### Steps to reproduce issue
```
colcon build \
  --symlink-install \
  --merge-install \
  --event-handlers console_cohesion+ console_package_list+ \
  --packages-skip-by-dep python_qt_binding \
  --cmake-args \
    --no-warn-unused-cli \
    -DBUILD_TESTING=OFF \
    -DINSTALL_EXAMPLES=ON \
    -DCMAKE_OSX_SYSROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk \
    -DCMAKE_OSX_ARCHITECTURES="x86_64" \
    -DCMAKE_PREFIX_PATH=$(brew --prefix):$(brew --prefix qt@5)``` 
```

```

#### Expected behavior
Installed Done
#### Actual behavior
Failed

clalancette, 2021-08-11:
It looks like you have 2 different versions of Qt installed, one in `/usr/local/include` and one in `/usr/local/opt/qt@5/lib`.

I have no idea if this is standard in Big Sur or not, but I'll suggest just having one installed.

ParthGaneriwala, 2021-08-11:
I just checked, there is no Qt in the `/usr/local/include` but there are 2 versions of Qt (Qt5 and Qt6). That happened after a normal installation of Qt.

ParthGaneriwala, 2021-08-11:
> It looks like you have 2 different versions of Qt installed, one in `/usr/local/include` and one in `/usr/local/opt/qt@5/lib`.
> 
> I have no idea if this is standard in Big Sur or not, but I'll suggest just having one installed.

I just checked, there is no Qt in the `/usr/local/include` but there are 2 versions of Qt (Qt5 and Qt6). That happened after a normal installation of Qt.

scott-robotics, 2024-09-19:
``` bash
brew remove Qt@6
brew install Qt@5
brew link Qt@5
```

Even though cmake finds Qt5, for some reason it uses the Qt6 qmake.