ros2/rviz #752: Point clouds are not cleared on time jump

Kettenhoax, 2021-08-25:
I'm using galactic with the released binary.

When displaying a point cloud topic during a time jump, for example when looping a ROS bag, subsequent point clouds stack until the original time point is reached again.
The decay time is disabled, i.e. only one cloud should be displayed.

[image]

Looking at the point cloud Display implementation on the current master, I've verified it's sufficient to decay clouds with receive time in the future in
`PointCloudCommon::cloudInfoIsDecayed`. However, I'm not sure if that's desirable in all cases. Alternatively the Display could install a ROS clock jump handler that clears data on jump.

```cpp
if (now < cloud_info->receive_time_) {
  return true;
}
```

I'm happy to provide a PR if a core team member could comment on the preferred solution. I've attached the [test data publisher and Rviz config](https://github.com/ros2/rviz/files/7045382/point_decay_time_jump.zip) to obtain the situation in the GIF.

gbalke, 2021-10-08:
Heya! I just got around to this and reproduced the issue locally with sim time:

```
rviz2 -d point_decay_time_jump.rviz --ros-args -p use_sim_time:=true
```

To note for later: The "Detected jump back in time" that's implemented in Rolling isn't in Foxy. Foxy never clears the TF buffer.

```
[WARN] [1633651498.409042849] [tf2_buffer]: Detected jump back in time. Clearing TF buffer.
```

My confusion is that, before the first TF buffer clear, the system operates as expected. Also hitting the "Reset" button seems to fix this state. It may be more reasonable to do a full reset on clock regression as we likely don't want any persistent state! I could be wrong here in the case of static objects that are loaded in at the beginning of a recording? @jacobperron any thoughts?

jacobperron, 2022-04-25:
Fixed in #791 

Proposing to backport for Humble in #854.

closed 2022-04-25 by jacobperron