ros2/rviz #762: build master fails rviz_rendering

adamkrawczyk, 2021-09-01:
Hi, 

ROS Foxy (binary install)

Build from source (master)

Ubuntu 20.04 

AMD64

When I followed [build tutorial](https://github.com/ros2/rviz#building-rviz-together-with-ros-2) I run into some troubles. 

First I removed rviz2 by `apt remove ros-foxy-rviz2` then:

`colcon build --merge-install`

```
--- stderr: rviz_rendering                                    
CMake Deprecation Warning at /opt/ros/foxy/src/gmock_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /opt/ros/foxy/src/gtest_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

/home/adam/ros2_ws/src/rviz/rviz_rendering/src/rviz_rendering/mesh_loader.cpp:65:10: fatal error: resource_retriever/retriever.hpp: No such file or directory
   65 | #include "resource_retriever/retriever.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rviz_rendering.dir/build.make:164: CMakeFiles/rviz_rendering.dir/src/rviz_rendering/mesh_loader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:338: CMakeFiles/rviz_rendering.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< rviz_rendering [8.22s, exited with code 2]

Summary: 2 packages finished [11min 47s]
  1 package failed: rviz_rendering
  1 package had stderr output: rviz_rendering
  5 packages not processed
```

Will try whether I can reproduce this error in docker container

clalancette, 2021-09-02:
Right, you are not going to be able to build the master branch against Foxy.  To build this repository, you either need to build `master` against Rolling, or you need to build the `foxy` branch against Foxy.

closed 2021-09-02 by clalancette

REGATTE, 2022-04-22:
HI,
I'm getting the same issue as above.

Ubuntu 20.04.04 LTS

Checked out the `Foxy` Branch

Followed the build tutorial, and still end up with the same issue. Any way to fix it?

Leaveszy, 2022-09-11:
Have you solved it? i have the same problem as you

commit 2ce58910 (2024-04-22): Merge pull request #764 from vrabaud/indigo-devel

fix #762