ros2/rviz #763: Missing MeshShape object

spurnvoj, 2021-09-08:
Hi everyone, I am just trying to migrate our rviz plugins from ROS 1 to ROS 2. I found out that compared to ROS 1 the RVIZ in ROS 2 doesn't have [MeshShape Object](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/rviz/ogre_helpers/mesh_shape.cpp). Is there any reason why this object hasn't been migrated yet?

clalancette, 2021-09-23:
Just lack of time, I would guess.  If you'd like to contribute it, we'd be happy to review.

ctoumieh, 2023-06-20:
Hey!
I am migrating a plugin from ROS to ROS2 that uses the MeshShape object. I was wondering if there was any progress on this issue or if @spurnvoj found a workaround.
Thanks.

spurnvoj, 2023-06-27:
Hi, It it some time back. I solved it by adding mesh source and header files into my repo and building it myself. Check the repo here: https://github.com/tiiuae/fog_rviz_plugins

ahcorde, 2023-09-14:
PR including MeshSahep https://github.com/ros2/rviz/pull/1064

ahcorde, 2023-11-10:
PR is merged, closing this for now.

closed 2023-11-10 by ahcorde