ros2/rviz #766: RVIz2 crashing

padhupradheep, 2021-09-15:
Distro: Rolling

**What am I doing?**
Trying to launch RViz2 from the terminal. 

**What is the error?** 
`[INFO] [1631706132.097429590] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1631706132.097503396] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1631706132.119974609] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1631706132.153348696] [rviz2]: PluginlibFactory: The plugin for class 'rviz_default_plugins/TF' failed to load. Error: Failed to load library /opt/ros/rolling/lib/librviz_default_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /opt/ros/rolling/lib/librviz_default_plugins.so: undefined symbol: _ZN7tf2_ros17TransformListener21subscription_callbackESt10shared_ptrIN8tf2_msgs3msg10TFMessage_ISaIvEEEEb, at /tmp/binarydeb/ros-rolling-rcutils-4.0.2/src/shared_library.c:99
[ERROR] [1631706132.194054541] [rviz2]: PluginlibFactory: The plugin for class 'rviz_default_plugins/Orbit' failed to load. Error: Failed to load library /opt/ros/rolling/lib/librviz_default_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /opt/ros/rolling/lib/librviz_default_plugins.so: undefined symbol: _ZN7tf2_ros17TransformListener21subscription_callbackESt10shared_ptrIN8tf2_msgs3msg10TFMessage_ISaIvEEEEb, at /tmp/binarydeb/ros-rolling-rcutils-4.0.2/src/shared_library.c:99
`

clalancette, 2021-09-15:
I can't seem to reproduce this problem locally; source `/opt/ros/rolling/setup.bash` and then running `rviz2` seems to work for me.  Running `dpkg -l | grep ros-rolling-rviz` shows me:

```
ii  ros-rolling-rviz-assimp-vendor                      8.7.0-1focal.20210811.162543         amd64        Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.
ii  ros-rolling-rviz-common                             8.7.0-1focal.20210902.145750         amd64        Common rviz API, used by rviz plugins and applications.
ii  ros-rolling-rviz-default-plugins                    8.7.0-1focal.20210902.170508         amd64        Several default plugins for rviz to cover the basic functionality.
ii  ros-rolling-rviz-ogre-vendor                        8.7.0-1focal.20210811.162606         amd64        Wrapper around ogre3d, it provides a fixed CMake module and an ExternalProject build of ogre.
ii  ros-rolling-rviz-rendering                          8.7.0-1focal.20210811.174355         amd64        Library which provides the 3D rendering functionality in rviz.
ii  ros-rolling-rviz2                                   8.7.0-1focal.20210902.175655         amd64        3D visualization tool for ROS.
```

What OS are you using, and what version of RViz2 are you running?

padhupradheep, 2021-09-15:
Ubuntu 20.04 and I have installed everything as binaries!

padhupradheep, 2021-09-15:
Oh okay, the problem is from my side. I think, I somehow override the default.rviz from other distro, which might have caused this issue. Once I deleted the default.rviz it works fine. But, I cannot give anyother suitable explanation on how this problem was caused.

closed 2021-09-15 by padhupradheep