ros2/rviz #773: Decreased performance in ros2-foxy versus ros2-dashing

simulacrus, 2021-09-23:
Hardware:
1. OS - ubuntu 20.04 / ubuntu 18.04
2. CPU - cpu intel core i7
3. GPU - Nvidia geforce gtx 250 (+proprietary driver)

Tests were carried out with the display of a large point cloud in version ros2-dashing (ubuntu 18) and version ros2-foxy (ubuntu 20). Statistics were collected on the duration of the renderOneFrame () function (file rviz2 / rviz_common / src / rviz_common / visualization_manager.cpp, line 437).

# ros2-dashing 
[image]

# ros2-foxy
[image]

Why is that?

gbalke, 2021-10-07:
I'm trying to debug this. Do you have a share-able bag file?

gbalke, 2021-10-13:
Since this is a bit of a long debug, I thought I should document my steps in case someone else wants to take a look (and for reference in future debugs).

I've used [uftrace](https://github.com/namhyung/uftrace) to follow all calls to `rviz_rendering::PointCloud*` using the following commands:

``` shell
# Build with instrumentation flags
$ colcon build --merge-install --cmake-args -DBUILD_TESTING:=OFF -DCMAKE_CXX_FLAGS:=-pg -DCMAKE_EXE_LINKER_FLAGS=-pg -DCMAKE_SHARED_LINKER_FLAGS=-pg
# Source install
$ source install/setup.sh
# Run uftrace (make sure you're going to be using the correct version of rviz with `which rviz2`)
$ uftrace record -F rviz_rendering::PointCloud* rviz2 -d ../../map.rviz
```

While the recorder is running, I run some replay from the bag. I found that rviz doesn't seem to register the single frame playback so I loop and exit as soon as rviz locks up (2-4 loops should do the trick).

```
ros2 bag play testmap/ -l
```

This locks up rviz for a minute... This isn't as bad when not instrumented. Eventually the pointcloud should appear:
[image]

At this point, close rviz and a `uftrace.data` folder should be in the current directory. Running `uftrace report` will generate the amount of time spent in each function call. This data will take a lot of time to accumulate as this is an absurd number of function calls that were recorded.

This in fact takes so long that it may not be worth checking (my trace broke 100GB when expanded to a json). In the hopes that this is not an issue caused at scale but instead individual paths, I reran with the point cloud generator made for #752 which generated a much more reasonably sized report.

``` shell
uftrace replay -F rviz_rendering::PointCloud::addPoints -t 1ms
# DURATION     TID     FUNCTION
            [126327] | rviz_rendering::PointCloud::addPoints() {
            [126327] |   std::vector::insert() {
            [126327] |     std::vector::_M_insert_dispatch() {
            [126327] |       std::vector::_M_range_insert() {
            [126327] |         std::__uninitialized_copy_a() {
            [126327] |           std::uninitialized_copy() {
   1.047 ms [126327] |             std::__uninitialized_copy::__uninit_copy();
   1.048 ms [126327] |           } /* std::uninitialized_copy */
   1.049 ms [126327] |         } /* std::__uninitialized_copy_a */
   1.065 ms [126327] |       } /* std::vector::_M_range_insert */
   1.067 ms [126327] |     } /* std::vector::_M_insert_dispatch */
   1.074 ms [126327] |   } /* std::vector::insert */
  25.191 ms [126327] | } /* rviz_rendering::PointCloud::addPoints */
```

The original test cloud is 1000 points so I doubled it to 2000 points and noticed these results:
``` shell
# DURATION     TID     FUNCTION
            [126481] | rviz_rendering::PointCloud::addPoints() {
            [126481] |   std::vector::insert() {
            [126481] |     std::vector::_M_insert_dispatch() {
            [126481] |       std::vector::_M_range_insert() {
            [126481] |         std::__uninitialized_copy_a() {
            [126481] |           std::uninitialized_copy() {
   1.557 ms [126481] |             std::__uninitialized_copy::__uninit_copy();
   1.558 ms [126481] |           } /* std::uninitialized_copy */
   1.559 ms [126481] |         } /* std::__uninitialized_copy_a */
   1.575 ms [126481] |       } /* std::vector::_M_range_insert */
   1.577 ms [126481] |     } /* std::vector::_M_insert_dispatch */
   1.583 ms [126481] |   } /* std::vector::insert */
  47.869 ms [126481] | } /* rviz_rendering::PointCloud::addPoints */
```

Digging a little deeper with `uftrace replay -F rviz_rendering::PointCloud::addPoints -D 3 -t 1us` I found this looping for what seems like forever. I assume this is an O(N) operation where N=number of points. Multiplying this out (~20us) by the 2k points gives us 40ms.
``` shell
          [126481] |   rviz_rendering::PointCloud::RenderableInternals::operator=() {
 1.563 us [126481] |     std::shared_ptr::operator=();
 1.643 us [126481] |   } /* rviz_rendering::PointCloud::RenderableInternals::operator= */
 1.853 us [126481] |   Ogre::AxisAlignedBox::merge();
16.231 us [126481] |   rviz_rendering::PointCloud::addPointToHardwareBuffer();
 ```
[... 32 lines cut ...]
   0.691 us [126481] |   } /* rviz_rendering::PointCloudRenderable::getBuffer */
            [126481] |   Ogre::SharedPtr::~SharedPtr() {
   0.511 us [126481] |     std::shared_ptr::~shared_ptr();
   0.581 us [126481] |   } /* Ogre::SharedPtr::~SharedPtr */
            [126481] |   rviz_rendering::PointCloudRenderable::getBuffer() {
            [126481] |     Ogre::SharedPtr::SharedPtr() {
   0.541 us [126481] |       std::shared_ptr::shared_ptr();
   0.611 us [126481] |     } /* Ogre::SharedPtr::SharedPtr */
   0.691 us [126481] |   } /* rviz_rendering::PointCloudRenderable::getBuffer */
            [126481] |   Ogre::SharedPtr::~SharedPtr() {
   0.510 us [126481] |     std::shared_ptr::~shared_ptr();
   0.582 us [126481] |   } /* Ogre::SharedPtr::~SharedPtr */
            [126481] |   rviz_rendering::PointCloudRenderable::getBuffer() {
            [126481] |     Ogre::SharedPtr::SharedPtr() {
   0.531 us [126481] |       std::shared_ptr::shared_ptr();
   0.612 us [126481] |     } /* Ogre::SharedPtr::SharedPtr */
   0.692 us [126481] |   } /* rviz_rendering::PointCloudRenderable::getBuffer */
            [126481] |   Ogre::SharedPtr::~SharedPtr() {
   0.511 us [126481] |     std::shared_ptr::~shared_ptr();
   0.581 us [126481] |   } /* Ogre::SharedPtr::~SharedPtr */
   0.511 us [126481] |   rviz_rendering::PointCloud::RenderableInternals::RenderableInternals();
  16.421 us [126481] | } /* rviz_rendering::PointCloud::addPointToHardwareBuffer */
```

There appear to be numerous creations/deletions of Ogre SharedPtr instances. These seem to be adding up to a whole lot of compute time. I'll dive in to the code tomorrow and also run the same analysis on Dashing.

gbalke, 2021-10-13:
I managed to reduce this time in half by moving the `getBuffer()` out of the assert and previous to the for loop. This, however, does not particularly improve the scenario as this is a different path from what we expect when running in release mode (assert is not run in release). If I change my build to:
```
colcon build --merge-install --cmake-args -DBUILD_TESTING:=OFF -DCMAKE_CXX_FLAGS:=-pg -DCMAKE_EXE_LINKER_FLAGS=-pg -DCMAKE_SHARED_LINKER_FLAGS=-pg -DCMAKE_BUILD_TYPE=Release
```
The execution of AddPoints drops down to 5ms for a 2000 point point cloud (not accounting for overhead). For 100k points, adding the points takes about 100ms.

@simulacrus at this point, it would be good to know if you're running on a release of rviz or if you're running a local build without the release flag? I'll be digging into Dashing while waiting for a response.

jcaraban, 2021-10-15:
@gbalke I believe the rviz2 foxy releases are not compiled with optimizations...
I cloned and compiled the foxy branch with optimizations, and it runs 2~3x faster in my use case and machine.

gbalke, 2021-10-15:
> I cloned and compiled the foxy branch with optimizations, and it runs 2~3x faster in my use case and machine.

Which optimizations? -O flags or the release flag?

simulacrus, 2021-10-18:
> > So...I try to turn off Selectable option on pointcloud2 plugin and ... it work. Stable 60+ fps. Why is that?
> 
> I disabled selectable and am still getting 0-1 fps: [image] I have no idea what's going on here... Are you somehow selecting points without realizing it?

Nope. Try to select Style Point and turn off autocompute option.

gbalke, 2021-10-18:
> Nope. Try to select Style Point and turn off autocompute option.

Same results with the following configuration:
[image]

gbalke, 2021-11-04:
@simulacrus I'm going to close this issue as a duplicate of #670. If you have additional information to add, please feel free to respond and the issue can be re-opened.

closed 2021-11-04 by gbalke