ros2/rviz #776: Noisy console output when clicking on render panel

jacobperron, 2021-09-30:
If you start RViz and click in the small space between the 3D render panel and the left panel divider, then we can see a bunch of print statements happening in the console, e.g.:

```
in RenderPanel::mousePressEvent()
in RenderPanel::mouseReleaseEvent()
in RenderPanel::mousePressEvent()
in RenderPanel::mouseReleaseEvent()
in RenderPanel::mouseDoubleClickEvent()
in RenderPanel::mouseReleaseEvent()
in RenderPanel::mousePressEvent()
in RenderPanel::mouseReleaseEvent()
in RenderPanel::mouseDoubleClickEvent()
in RenderPanel::mouseReleaseEvent()
in RenderPanel::mousePressEvent()
in RenderPanel::mouseMoveEvent()
in RenderPanel::mouseMoveEvent()
```

It'd be nice to remove these noisy prints.

jacobperron, 2021-09-30:
Here's an example of where you can click to reproduce the issue:

[image]

ahcorde, 2021-10-01:
https://github.com/ros2/rviz/pull/777

closed 2021-10-04 by ahcorde