ros2/rviz #782: PointCloud2 and Other Topics no longer Display in RVIZ after upgrade from FOXY Release 5 to FOXY Release 6

tomcat327, 2021-10-08:
Today, I "apt upgraded" to FOXY Patch Release 6, which came out on Oct. 7, 2021. Previously, I had been on FOXY Patch Release 5, which came out on Sept. 9, 2021. See this link for patch release details: https://github.com/ros2/ros2/releases.

For over a year, I have been using an rviz configuration file ([ouster_live.zip](https://github.com/ros2/rviz/files/7313693/ouster_live.zip)) with RVIZ2 to watch PointCloud2s coming from "ros2 bag play". This is all managed within a launch file.

However, after the upgrade to 6, I can no longer visualize PointCloud2s in RVIZ2. 

Here is a PointCloud2-less screenshot showing no point clouds on the left and the results of "ros2 topic info -v /MAP_LiDAR/PointCloud2" on the right:
[image]

The result of "ros2 topic echo /MAP_LiDAR/PointCloud2" is shown below, proving that PointCloud2s are definitely being published:
[image]

For this issue, I'm focusing on PointCloud2s, but I am also having the same problem with visualization of Path and Pose topics. This seems to be a bug, but if it is user error, I apologize in advance. Something seems to have changed (external to my unchanged setup) in the last month that is causing these issues. Thank you in advance!

gbalke, 2021-10-08:
Verified. I'll start taking a look. @jacobperron any guesses?

jacobperron, 2021-10-08:
@gbalke I would start by doing a bisect between 8.2.3 and 8.2.4 to find an offending commit: https://github.com/ros2/rviz/commits/foxy

gbalke, 2021-10-08:
@tomcat327 fix is on the way :smiley:

tomcat327, 2021-10-08:
Thank you for the fast response @gbalke!

gbalke, 2021-10-08:
Root caused the problem... Pretty interesting tbh :thinking:
https://github.com/ros2/rviz/pull/783#issuecomment-939139321

kallaspriit, 2021-10-09:
Can also confirm this, pulling out my hair why stuff broke that was working yesterday without changing anything :)

How could one revert to the previous version while this gets fixed? I'm using `ros-foxy-rviz2` package on Ubuntu 20.04.

```
$ apt info ros-foxy-rviz2
Package: ros-foxy-rviz2
Version: 8.2.4-1focal.20211006.001624
Priority: optional
Section: misc
Maintainer: Jacob Perron <jacob@openrobotics.org>
Installed-Size: 92.2 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libqt5core5a (>= 5.12.2), libqt5widgets5 (>= 5.0.2), libstdc++6 (>= 5.2), ros-foxy-rviz-common, ros-foxy-rviz-default-plugins, ros-foxy-rviz-ogre-vendor, ros-foxy-ros-workspace
Homepage: https://github.com/ros2/rviz/blob/ros2/README.md
Download-Size: 12.6 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
Description: 3D visualization tool for ROS.
```

Tried listing old versions but it shows none, doing it wrong?
```
$ apt-cache madison ros-foxy-rviz2
ros-foxy-rviz2 | 8.2.4-1focal.20211006.001624 | http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
```

Tried cloning tag `8.2.3` into my workspace and `colcon build` it but the build fails.
```
[Processing: rviz_default_plugins]                                         
--- stderr: rviz_default_plugins                                           
/usr/bin/ld: CMakeFiles/point_cloud2_display_test.dir/test/rviz_default_plugins/displays/pointcloud/point_cloud2_display_test.cpp.o: in function `rviz_common::MessageFilterDisplay<sensor_msgs::msg::PointCloud2_<std::allocator<void> > >::incomingMessage(std::shared_ptr<sensor_msgs::msg::PointCloud2_<std::allocator<void> > const>)':
point_cloud2_display_test.cpp:(.text._ZN11rviz_common20MessageFilterDisplayIN11sensor_msgs3msg12PointCloud2_ISaIvEEEE15incomingMessageESt10shared_ptrIKS5_E[_ZN11rviz_common20MessageFilterDisplayIN11sensor_msgs3msg12PointCloud2_ISaIvEEEE15incomingMessageESt10shared_ptrIKS5_E]+0x59): undefined reference to `rviz_common::_RosTopicDisplay::typeErasedMessageTaken(std::shared_ptr<void const>)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/point_cloud2_display_test.dir/build.make:254: point_cloud2_display_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: CMakeFiles/point_cloud2_display_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/point_cloud_scalar_display_test.dir/test/rviz_default_plugins/displays/pointcloud/point_cloud_scalar_display_test.cpp.o: in function `rviz_common::MessageFilterDisplay<sensor_msgs::msg::Illuminance_<std::allocator<void> > >::incomingMessage(std::shared_ptr<sensor_msgs::msg::Illuminance_<std::allocator<void> > const>)':
point_cloud_scalar_display_test.cpp:(.text._ZN11rviz_common20MessageFilterDisplayIN11sensor_msgs3msg12Illuminance_ISaIvEEEE15incomingMessageESt10shared_ptrIKS5_E[_ZN11rviz_common20MessageFilterDisplayIN11sensor_msgs3msg12Illuminance_ISaIvEEEE15incomingMessageESt10shared_ptrIKS5_E]+0x59): undefined reference to `rviz_common::_RosTopicDisplay::typeErasedMessageTaken(std::shared_ptr<void const>)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/point_cloud_scalar_display_test.dir/build.make:238: point_cloud_scalar_display_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:584: CMakeFiles/point_cloud_scalar_display_test.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< rviz_default_plugins [2min 20s, exited with code 2]
```
So how do I revert to the working version or just have to wait for the fix to land in `ros-foxy-rviz2`?

fantauzzi, 2021-10-10:
I think I am having the same issue in Rviz2 with /scan and /map since the update to patch release 6: Rviz2 doesn't show the LaserScan from a Turtblebot 3 (real or simulated in Gazebo), nor the map being built by cartographer.
[image]

Paul-Ajay, 2021-10-11:
Any temporary solution available? until the release of binaries with fix?

gavanderhoorn, 2021-10-12:
> Actually getting this into a release will take some time given the length of the process

this seems like a serious enough regression to warrant an out-of-band patch release?

closed 2021-10-13 by jacobperron