ros2/rviz #796: After `sudo apt upgrade`, several rviz_default_plugins fail to load

AndyZe, 2021-11-07:
On ROS2 Rolling, I just did `sudo apt update/upgrade`. Now I get these messages when launching rviz2:

> [rviz2-1] [ERROR] [1636328709.608941623] [rviz2]: PluginlibFactory: The plugin for class 'rviz_default_plugins/TF' failed to load. Error: Failed to load library /opt/ros/rolling/lib/librviz_default_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /opt/ros/rolling/lib/librviz_default_plugins.so: undefined symbol: _ZN7tf2_ros17TransformListener21subscription_callbackESt10shared_ptrIKN8tf2_msgs3msg10TFMessage_ISaIvEEEEb, at /tmp/binarydeb/ros-rolling-rcutils-5.0.0/src/shared_library.c:99

> [rviz2-1] [ERROR] [1636328709.648015691] [rviz2]: PluginlibFactory: The plugin for class 'rviz_default_plugins/Orbit' failed to load. Error: Failed to load library /opt/ros/rolling/lib/librviz_default_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /opt/ros/rolling/lib/librviz_default_plugins.so: undefined symbol: _ZN7tf2_ros17TransformListener21subscription_callbackESt10shared_ptrIKN8tf2_msgs3msg10TFMessage_ISaIvEEEEb, at /tmp/binarydeb/ros-rolling-rcutils-5.0.0/src/shared_library.c:99

AndyZe, 2021-11-07:
`apt-cache policy ros-rolling-rviz-default-plugins` shows:

```
ros-rolling-rviz-default-plugins:
  Installed: 8.7.0-1focal.20211102.045605
  Candidate: 8.7.0-1focal.20211102.045605
  Version table:
 *** 8.7.0-1focal.20211102.045605 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

`apt-cache policy ros-rolling-rviz2` shows:

```
  Installed: 8.7.0-1focal.20211102.060509
  Candidate: 8.7.0-1focal.20211102.060509
  Version table:
 *** 8.7.0-1focal.20211102.060509 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

clalancette, 2021-11-08:
I can't reproduce this here; rviz2 launches fine for me.  What version of `ros-rolling-tf2-ros` do you have?  I have `0.19.0-1focal.20211101.215611`.

AndyZe, 2021-11-08:
I have the same version

AndyZe, 2021-11-08:
Ah, it works fine if I don't source my workspace but simply source `/opt/ros/rolling/setup.bash`. OK, I guess it's a problem with my workspace, then.

closed 2021-11-08 by AndyZe

AndyZe, 2021-11-08:
This seemed to fix it for me. I don't understand why. Probably related to `tf2_msgs` since that was mentioned in the error printout.

```
sudo apt purge python-is-python2
sudo apt install python3-vcstools python3-wstool
```