ros2/rviz #797: Plugin use_sim_time has no effect

crabiner, 2021-11-08:
I have set "use_sim_time" to true on my custom panel for rviz but the clock is still machine clock and not simulated time
I have tried setting the flag on runtime as well, and in callback loop I always see machine clock

see snippet below
```
ControlPanel::ControlPanel(QWidget* parent) : Panel(parent) 
{
    // Setup ROS2 functionality
    node_ = rclcpp::Node::make_shared("control_panel");
    node_->set_parameter(rclcpp::Parameter("use_sim_time", true));

    auto now = node_->get_clock()->now();
    RCLCPP_INFO(node_->get_logger(), std::to_string(now.nanoseconds()).c_str());

}

dmorris0, 2022-09-26:
I have the same problem.  I tried setting `use_sim_time` with the command line:
  `ros2 param set /rviz use_sim_time true`
but rviz still does not use the time in the rosbag being played

ufrhaidar, 2023-04-18:
you're setting the parameter after creating the node, so you'll have to spin a couple of times before the node reads the clock topic properly. Preferably add the parameters to rclcpp::NodeOptions and start the node with those options.