ros2/rviz #799: Various rviz2 segfaults during live navigation2 runs

bergercookie, 2021-11-10:
I 'm using rviz2 to visualise the navigation2 process for an actual (non-simulation) robot.
There are many occasions where rviz2 crashes out of the blue. I believe most of these crashes relate to the visualisation of the local costmap of nav2 since I have not seen rviz2 crash when the local costmap viz group is disabled (see attached picture for the said viualisation group)

I'm using ROS2 Foxy, and here are the installed versions of the rviz2 related packages:

```
$ dpkg -l | grep rviz
ii  ros-foxy-nav2-rviz-plugins                      0.4.7-1focal.20211012.200747         amd64        Navigation 2 plugins for rviz
ii  ros-foxy-rviz-assimp-vendor                     8.2.5-1focal.20211012.183928         amd64        Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.
ii  ros-foxy-rviz-common                            8.2.5-1focal.20211012.190724         amd64        Common rviz API, used by rviz plugins and applications.
ii  ros-foxy-rviz-default-plugins                   8.2.5-1focal.20211012.192708         amd64        Several default plugins for rviz to cover the basic functionality.
ii  ros-foxy-rviz-ogre-vendor                       8.2.5-1focal.20211012.183909         amd64        Wrapper around ogre3d, it provides a fixed CMake module and an ExternalProject build of ogre.
ii  ros-foxy-rviz-ogre-vendor-dbgsym                8.2.5-1focal.20211012.183909         amd64        debug symbols for ros-foxy-rviz-ogre-vendor
ii  ros-foxy-rviz-rendering                         8.2.5-1focal.20211012.190300         amd64        Library which provides the 3D rendering functionality in rviz.
ii  ros-foxy-rviz2                                  8.2.5-1focal.20211012.200723         amd64        3D visualization tool for ROS.
```

I have installed the debugging symbol packages for ogre and rviz so I can produce meaningful backtraces and core dumps if that's helpful in debugging. I'm also attaching a sample backtrace and a view of threads at the time of one of these crashes:

```
(gdb) i threads
  Id   Target Id                                            Frame
* 1    Thread 0x7ffff0fa21c0 (LWP 197655) "rviz2"           0x00007fffe43e76e0 in Ogre::VertexElement::getSource() const () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
  2    Thread 0x7ffff016c700 (LWP 197685) "QXcbEventQueue"  0x00007ffff6d18aff in poll () from /usr/lib/x86_64-linux-gnu/libc.so.6
  3    Thread 0x7fffef010700 (LWP 197686) "rviz2"           0x00007ffff59963f4 in do_futex_wait.constprop () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  4    Thread 0x7fffee786700 (LWP 197687) "rviz2"           0x00007ffff59937b1 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  5    Thread 0x7fffedf85700 (LWP 197688) "rviz2"           0x00007ffff6d2661d in recvmsg () from /usr/lib/x86_64-linux-gnu/libc.so.6
  6    Thread 0x7fffed784700 (LWP 197689) "rviz2"           0x00007ffff6d2661d in recvmsg () from /usr/lib/x86_64-linux-gnu/libc.so.6
  7    Thread 0x7fffecf79700 (LWP 197690) "rviz2"           0x00007ffff59937b1 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  8    Thread 0x7fffe7fff700 (LWP 197691) "rviz2"           0x00007ffff5996618 in do_futex_wait.constprop () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  9    Thread 0x7fffe77fe700 (LWP 197692) "rviz2"           0x00007ffff6d2661d in recvmsg () from /usr/lib/x86_64-linux-gnu/libc.so.6
  10   Thread 0x7fffe6ffd700 (LWP 197693) "rviz2"           0x00007ffff5996618 in do_futex_wait.constprop () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  12   Thread 0x7fffe5ffb700 (LWP 197695) "rviz2"           0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  13   Thread 0x7fffe57fa700 (LWP 197696) "QDBusConnection" 0x00007ffff6d18aff in poll () from /usr/lib/x86_64-linux-gnu/libc.so.6
  14   Thread 0x7fffc0f35700 (LWP 197700) "rviz2:disk$0"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  15   Thread 0x7fffc0734700 (LWP 197701) "rviz2:disk$1"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  16   Thread 0x7fffbff33700 (LWP 197702) "rviz2:disk$2"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  17   Thread 0x7fffbf732700 (LWP 197703) "rviz2:disk$3"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  18   Thread 0x7fffbef31700 (LWP 197704) "rviz2:disk$0"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  19   Thread 0x7fffbe730700 (LWP 197705) "rviz2:disk$1"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  20   Thread 0x7fffbdf2f700 (LWP 197706) "rviz2:disk$2"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  21   Thread 0x7fffbd72e700 (LWP 197707) "rviz2:disk$3"    0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  24   Thread 0x7fff9f3b2700 (LWP 197710) "rviz2"           0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  25   Thread 0x7fffe67fc700 (LWP 197711) "rviz2"           0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0
  27   Thread 0x7fff9e998700 (LWP 197713) "nav2_rviz_plugi" 0x00007ffff5993376 in pthread_cond_wait@@GLIBC_2.3.2 () from /usr/lib/x86_64-linux-gnu/libpthread.so.0

(gdb) thread 1
[Switching to thread 1 (Thread 0x7ffff0fa21c0 (LWP 197655))]
#0  0x00007fffe43e76e0 in Ogre::VertexElement::getSource() const () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1

(gdb) bt
#0  0x00007fffe43e76e0 in Ogre::VertexElement::getSource() const () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
#1  0x00007fffe43e21ee in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
#2  0x00007ffff689dd09 in Ogre::SceneManager::_issueRenderOp(Ogre::Renderable*, Ogre::Pass const*) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#3  0x00007ffff689450d in Ogre::SceneManager::issueRenderWithLights(Ogre::Renderable*, Ogre::Pass const*, Ogre::HashedVector<Ogre::Light*> const*, bool) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#4  0x00007ffff6895479 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#5  0x00007ffff6893c29 in Ogre::SceneManager::SceneMgrQueuedRenderableVisitor::visit(Ogre::Pass const*, std::vector<Ogre::Renderable*, std::allocator<Ogre::Renderable*> >&) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#6  0x00007ffff68001e9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#7  0x00007ffff6800074 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#8  0x00007ffff6894016 in Ogre::SceneManager::renderObjects(Ogre::QueuedRenderableCollection const&, Ogre::QueuedRenderableCollection::OrganisationMode, bool, bool, Ogre::HashedVector<Ogre::Light*> const*, bool) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#9  0x00007ffff6894254 in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#10 0x00007ffff68941a2 in Ogre::SceneManager::_renderQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#11 0x00007ffff6893a86 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () from /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
[... cut ...]
#35 0x00007ffff74bb565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff74624db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00007ffff746a246 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x0000555555556a96 in ?? ()
#39 0x00007ffff6c2a0b3 in __libc_start_main () from /usr/lib/x86_64-linux-gnu/libc.so.6
#40 0x0000555555556b5e in ?? ()
```

Let me know what other information may be helpful to debug this issue; I'm happy to provide it.

bergercookie, 2021-11-10:
Probably similar  to #703 .

clalancette, 2022-03-02:
Since we've now solved #703 , I have hope that this is solved as well.  @bergercookie can you give the latest Rolling sources a try and see if this is any better for you?

clalancette, 2023-07-14:
Given that we believe this has been fixed in later versions of RViz, and that Foxy is now End-of-Life, I'm going to close this out.  Please feel free to open a new issue if you are still encountering this problem.

closed 2023-07-14 by clalancette