ros2/rviz #804: Can't see robot model using WSL2 ROS Galactic

sink0918, 2021-11-27:
`rviz2` does not have any error outputs.
[image]

can't see robot model in rviz2:
[image]

iuhilnehc-ynos, 2021-11-29:
There is `a robot model` in the pic.
[image]

but I know that it might be not what you want, could you check your robot model description file?  (or you can upload it if someone wants to help.)

sink0918, 2021-11-29:
@iuhilnehc-ynos This is the collision, not the visual mesh model. Sorry that I forgot to update the expected behavior.

sink0918, 2021-11-30:
Without using stl it works fine:
[image]

The problem seems to be "stl meshes" can not be rendered in rviz2 using WSL2.

By the way, I'm using `WSLg` to use `rviz2`, not `X11-forwarding`.

And I also tried [realsense-description](https://github.com/IntelRealSense/realsense-ros/tree/ros2/realsense2_description), the stl mesh can't be seen either.

iuhilnehc-ynos, 2021-11-30:
@sink0918 

Thank you for your information, but I'm sorry that I can't reproduce it. (I use windows10 so that I can't use WSLg.)

[image]

sink0918, 2021-12-01:
[image]

I have tested it using X11-forwarding, but the mesh still can't be rendered😢

Here is some infos of my WSL2:
- Windows Build Number
Microsoft Windows [版本 10.0.22000.194]

- WSL Version
 WSL 2

- Kernel Version
5.10.60.1

sink0918, 2021-12-01:
I have also reported this issue to microsoft/WSL, here is the link: https://github.com/microsoft/WSL/issues/7788

dpiet, 2021-12-21:
I can confirm experiencing this issue as well.

ravnicas, 2022-01-26:
+1 here, tried on Win11 with iGPU and also nVidia GPU systems.

dpiet, 2022-04-09:
This appears to still be an issue in rolling, rviz2 v9.1.1. Any ideas what could be the culprit?

junaidamin, 2023-02-24:
Did anyone find a solution to this issue? I am using ROS2 foxy on WSL2 Ubuntu 20.04, and could not see a rendering of either the dae file or stl files. There are no related errors in the terminal. 
[image]

junaidamin, 2023-02-24:
Also, does anyone know why the sphere (head of robot) surface isn't smooth?

chenjunnn, 2023-03-02:
It works well with ROS2 Humble with WSL Ubuntu 22.04 now.

closed 2023-03-02 by chenjunnn