ros2/rviz #810: ItemIdentityException: Can not find codec for 'tif' image format.

andreasBihlmaier, 2021-12-23:
On a clean apt package install of galactic on Ubuntu 20.04.3, it seems that Ogre has been compiled without TIF support:

```
[rviz2-3] [INFO] [1640255530.859794173] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1640255530.859999453] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[joint_state_publisher_gui-1] [INFO] [1640255530.862830324] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-1] [INFO] [1640255530.870288294] [joint_state_publisher]: Centering
[rviz2-3] [INFO] [1640255531.419015282] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-1] [INFO] [1640255531.433760590] [joint_state_publisher]: Centering
[rviz2-3] [ERROR] [1640255531.703707234] [rviz2]: ItemIdentityException: Can not find codec for 'tif' image format.
[rviz2-3] Supported formats are: bmp dds gif hdr jpeg jpg ktx pgm pic pkm png ppm psd tga. in Codec::getCodec at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreCodec.cpp (line 63)
[rviz2-3] [ERROR] [1640255531.703783520] [rviz2]: Could not load texture [package://urdf_tutorial/meshes/l_finger_color.tif]: ItemIdentityException: Can not find codec for 'tif' image format.
[rviz2-3] Supported formats are: bmp dds gif hdr jpeg jpg ktx pgm pic pkm png ppm psd tga. in Codec::getCodec at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreCodec.cpp (line 63)
[rviz2-3] [ERROR] [1640255531.705266120] [rviz2]: FileNotFoundException: Cannot locate resource package://urdf_tutorial/meshes/l_finger_color.tif in resource group rviz_rendering. in ResourceGroupManager::openResource at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
[rviz2-3] [ERROR] [1640255531.705339447] [rviz2]: Error: preparing texture 'package://urdf_tutorial/meshes/l_finger_color.tif'. Texture layer will be blank: Cannot locate resource package://urdf_tutorial/meshes/l_finger_color.tif in resource group rviz_rendering.
[rviz2-3] [ERROR] [1640255531.705354173] [rviz2]: FileNotFoundException: Cannot locate resource package://urdf_tutorial/meshes/l_finger_color.tif in resource group rviz_rendering. in ResourceGroupManager::openResource at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
[rviz2-3] [ERROR] [1640255531.705373260] [rviz2]: Error: preparing texture 'package://urdf_tutorial/meshes/l_finger_color.tif'. Texture layer will be blank: Cannot locate resource package://urdf_tutorial/meshes/l_finger_color.tif in resource group rviz_rendering.
[rviz2-3] [ERROR] [1640255531.706538434] [rviz2]: ItemIdentityException: Can not find codec for 'tif' image format.
[rviz2-3] Supported formats are: bmp dds gif hdr jpeg jpg ktx pgm pic pkm png ppm psd tga. in Codec::getCodec at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreCodec.cpp (line 63)
[rviz2-3] [ERROR] [1640255531.706572149] [rviz2]: Could not load texture [package://urdf_tutorial/meshes/l_finger_tip_color.tif]: ItemIdentityException: Can not find codec for 'tif' image format.
[rviz2-3] Supported formats are: bmp dds gif hdr jpeg jpg ktx pgm pic pkm png ppm psd tga. in Codec::getCodec at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreCodec.cpp (line 63)
[rviz2-3] [ERROR] [1640255531.707129598] [rviz2]: FileNotFoundException: Cannot locate resource package://urdf_tutorial/meshes/l_finger_tip_color.tif in resource group rviz_rendering. in ResourceGroupManager::openResource at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
[rviz2-3] [ERROR] [1640255531.707167703] [rviz2]: Error: preparing texture 'package://urdf_tutorial/meshes/l_finger_tip_color.tif'. Texture layer will be blank: Cannot locate resource package://urdf_tutorial/meshes/l_finger_tip_color.tif in resource group rviz_rendering.
[rviz2-3] [ERROR] [1640255531.707178508] [rviz2]: FileNotFoundException: Cannot locate resource package://urdf_tutorial/meshes/l_finger_tip_color.tif in resource group rviz_rendering. in ResourceGroupManager::openResource at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.0/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
[rviz2-3] [ERROR] [1640255531.707196495] [rviz2]: Error: preparing texture 'package://urdf_tutorial/meshes/l_finger_tip_color.tif'. Texture layer will be blank: Cannot locate resource package://urdf_tutorial/meshes/l_finger_tip_color.tif in resource group rviz_rendering.
```

This error shows up when trying to run the official URDF tutorial: http://docs.ros.org/en/galactic/Tutorials/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html#finishing-the-model due to the finger meshes.
Specifically when running
```
ros2 launch urdf_tutorial display.launch.py model:=urdf/05-visual.urdf
```

urdf_tutorial is a fresh checkout of https://github.com/ros/urdf_tutorial/tree/ros2 on ros2 branch, i.e. f8a1bf7f01c61bdbfc9a1cd25f7dda3843af0a0d.

clalancette, 2022-01-07:
I took a bit of a look at what was going on here today.

First of all, I was able to reproduce the problem very easily, even on Rolling and even building from source.  As the error message says, the problem is that `tif` (or `tiff`) files aren't currently supported by `rviz_ogre_vendor`.  The obvious workaround is to change the `tif` files in `urdf_tutorial` to another file format that we *can* successfully load.

But that is a bit unsatisfying; it would be nice to support `tif` files.

So I took a deeper look.  What happens when Ogre starts up is that it has plugins for handling various file image types.  One of those plugins is for freeimage, and one of the plugins is for STBI.  freeimage is the well-known open-source library, and does seem to support `tif` files.  STBI seems to be a relatively simple image library with support for various image file formats, but not for `tif`.  Crucially, STBI is bundled into the Ogre sources, while freeimage is only found via an external library.  So at build time, if freeimage is found, then Ogre builds the freeimage plugin.  STBI is always in the sources so it is always built.

And this leads us to the first problem.  Our rviz_ogre_vendor [package.xml]() file does *not* declare a dependency on `freeimage`.  What that means is that when we are [building the packages on the buildfarm](https://github.com/ros2/rviz/blob/d084db174db0ca4170759c0d9c50b49f54b4a7f9/rviz_ogre_vendor/package.xml), it does not have access to `freeimage` (since we only install packages in the package.xml).  This in turn means that the packages don't have support for freeimage, and thus will never support `tif`.

That said, we should be able to workaround this locally by installing `libfreeimage-dev` ourselves, and then building the package.  Indeed, doing that we can see that the build does indeed build the freeimage plugin.  However, for reasons I don't yet understand, it still does not use freeimage and still falls back to the STBI library.

That is the next thing to investigate here; once we figure out how Ogre chooses between the libraries, we should be able to (hopefully) make it use freeimage and get `tif` support.

xlla, 2022-05-10:
I have encounter this issue while study the official URDF tutorial.
my env is OSX 10.13 , ros 2 humble. Obviously it is ancient and out of official support,  so I have to spend many hours to solve this issue.

I found the rviz_ogre_vendor was built fine , Codec_FreeImage.1.12.1.dylib was deployed.
I changed opt/rviz_ogre_vendor/bin/plugins.cfg to enable Codec_FreeImage, the errors remain.
I have found another config file share/rviz_rendering/ogre_media/plugins.cfg, it seems just a blank file.
I finally deep into the source and found render_system.cpp does not load  Codec_FreeImage at all!
After manual add Codec_FreeImage and recompile the rviz_rendering, everything is work now.

<pre>
slim:urdf_tutorial xlla$ ros2 launch urdf_tutorial display.launch.py model:=urdf/06-flexible.urdf
[INFO] [launch]: All log files can be found below /Users/xlla/.ros/log/2022-05-11-01-56-45-771014-slim.local-92368
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher_gui-1]: process started with pid [92371]
[INFO] [robot_state_publisher-2]: process started with pid [92372]
[INFO] [rviz2-3]: process started with pid [92373]
[robot_state_publisher-2] Link left_leg had 1 children
[robot_state_publisher-2] Link left_base had 2 children
[robot_state_publisher-2] Link left_back_wheel had 0 children
[robot_state_publisher-2] Link left_front_wheel had 0 children
[robot_state_publisher-2] Link right_leg had 1 children
[robot_state_publisher-2] Link right_base had 2 children
[robot_state_publisher-2] Link right_back_wheel had 0 children
[robot_state_publisher-2] Link right_front_wheel had 0 children
[robot_state_publisher-2] Link gripper_pole had 2 children
[robot_state_publisher-2] Link left_gripper had 1 children
[robot_state_publisher-2] Link left_tip had 0 children
[robot_state_publisher-2] Link right_gripper had 1 children
[robot_state_publisher-2] Link right_tip had 0 children
[robot_state_publisher-2] Link head had 1 children
[robot_state_publisher-2] Link box had 0 children
[robot_state_publisher-2] [INFO] [1652205406.503559629] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1652205406.503686723] [robot_state_publisher]: got segment box
[robot_state_publisher-2] [INFO] [1652205406.503713519] [robot_state_publisher]: got segment gripper_pole
[robot_state_publisher-2] [INFO] [1652205406.503732872] [robot_state_publisher]: got segment head
[robot_state_publisher-2] [INFO] [1652205406.503754609] [robot_state_publisher]: got segment left_back_wheel
[robot_state_publisher-2] [INFO] [1652205406.503780609] [robot_state_publisher]: got segment left_base
[robot_state_publisher-2] [INFO] [1652205406.503807145] [robot_state_publisher]: got segment left_front_wheel
[robot_state_publisher-2] [INFO] [1652205406.503830548] [robot_state_publisher]: got segment left_gripper
[robot_state_publisher-2] [INFO] [1652205406.503855400] [robot_state_publisher]: got segment left_leg
[robot_state_publisher-2] [INFO] [1652205406.503878857] [robot_state_publisher]: got segment left_tip
[robot_state_publisher-2] [INFO] [1652205406.503902527] [robot_state_publisher]: got segment right_back_wheel
[robot_state_publisher-2] [INFO] [1652205406.503925159] [robot_state_publisher]: got segment right_base
[robot_state_publisher-2] [INFO] [1652205406.503945263] [robot_state_publisher]: got segment right_front_wheel
[robot_state_publisher-2] [INFO] [1652205406.503999806] [robot_state_publisher]: got segment right_gripper
[robot_state_publisher-2] [INFO] [1652205406.504054232] [robot_state_publisher]: got segment right_leg
[robot_state_publisher-2] [INFO] [1652205406.504084963] [robot_state_publisher]: got segment right_tip
[rviz2-3] [INFO] [1652205407.135175489] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1652205407.135370415] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[rviz2-3] [INFO] [1652205407.168600361] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-1] [INFO] [1652205407.362290239] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-1] [INFO] [1652205407.384668100] [joint_state_publisher]: Centering
[joint_state_publisher_gui-1] [INFO] [1652205407.601454869] [joint_state_publisher]: Centering
</pre>

I have created a PR #863