ros2/rviz #819: can't turn off the verbose or change the logging level.

sinamr, 2022-01-20:
Hey,

I am trying to run rviz without any verbose or logging but so far any solution I could find results in a failure. For example the following is not working:

```
ros2 run rviz2 rviz2 -ros-args --log-level FATAL
```
as it prints tons of warnings (which I legit, but I don't want to have the console being spammed); as an example :
```
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 332 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame  at time 1642693568.749882 according to authority Authority undetectable
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 332 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame  at time 1642693568.754049 according to authority Authority undetectable
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 332 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
CWarning: Invalid frame ID "" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.9/src/buffer_core.cpp
``` 

Any suggestion ?

tonynajjar, 2022-09-27:
This seems to work for me on galactic:

```
root@logi-XMG-CORE-REN-M20:/code# ros2 run rviz2 rviz2 --ros-args --log-level FATAL
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root
```

tonynajjar, 2022-09-27:
Mmm nope, my answer is not very useful, I also still get the warnings you were talking about. I guess these warnings are not printed by a "ros logger", so --log-level has no power. Correct me if I'm wrong.

alexmillane, 2022-10-11:
+1 on this issue.

Spamming the console with TF errors makes finding the actual source of issues difficult. This is particularly true for people maintaining libraries, because library users who hit issues, tend to then post snapshots of the console which contains only TF spam. Meaningful output from the system is drowned out.

From what I can tell there's no way to stop TF warnings coming from RVIZ. Is that correct?

leander2189, 2023-03-01:
I am having the same problem and came across with this issue. Is there any update on this?

jkaob, 2023-03-13:
I've been looking for a solution to this as well. I'd _really_ like to be able to disable all  outputs/logs when launching rviz together with other nodes

jamestkpoon, 2023-08-02:
Apologies for resurrecting an old thread, but this issue is one of the top results for e.g. "suppress rviz2 output.'
This [answer](https://answers.ros.org/question/349176/control-ros2-logging-output-with-launch-command/?answer=350304#post-id-350304) worked for me:
```
output={'both': 'log'}
```