ros2/rviz #832: rviz_rendering won't build because of CMake find_package issue

NoodleOfDeath, 2022-02-20:
When trying to build I get the following errors and Stackoverflow does not appear to have a solution:

```
$ colcon build --symlink-install
[0.880s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'create_robot' is in: /home/ubuntu/dev/install/create_robot
	'rplidar_ros2' is in: /home/ubuntu/dev/install/rplidar_ros2
	'rviz_assimp_vendor' is in: /home/ubuntu/dev/install/rviz_assimp_vendor
	'rviz_ogre_vendor' is in: /home/ubuntu/dev/install/rviz_ogre_vendor
	'rviz_rendering' is in: /home/ubuntu/dev/install/rviz_rendering
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding create_robot rplidar_ros2 rviz_assimp_vendor rviz_ogre_vendor rviz_rendering

This may be promoted to an error in a future release of colcon-core.
Starting >>> rviz_assimp_vendor
Starting >>> rviz_ogre_vendor
Starting >>> create_robot
Starting >>> rplidar_ros2
Finished <<< create_robot [3.35s]                                                                                 
Finished <<< rviz_assimp_vendor [3.42s]
Finished <<< rviz_ogre_vendor [3.39s]
Starting >>> rviz_rendering
Finished <<< rplidar_ros2 [5.31s]                                                                
--- stderr: rviz_rendering                          
CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "point_cloud_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:181 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "point_cloud_test_target" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:181 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "wrench_visual_test_target" links to target
  "rviz_ogre_vendor::OgreOverlay" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:266 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "wrench_visual_test_target" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:266 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "string_helper_test" links to target "rviz_ogre_vendor::OgreOverlay"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:176 (ament_add_gmock)
[... 132 lines cut ...]
  Target "line_test_target" links to target "rviz_ogre_vendor::OgreOverlay"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:254 (ament_add_gmock)

CMake Error at /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:69 (add_executable):
  Target "line_test_target" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:52 (_ament_add_gmock)
  CMakeLists.txt:254 (ament_add_gmock)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< rviz_rendering [2.57s, exited with code 1]

Summary: 4 packages finished [6.48s]
  1 package failed: rviz_rendering
  1 package had stderr output: rviz_rendering
  5 packages not processed
```

jacobperron, 2022-03-09:
I wonder if this is related to #601 
Could you try uninstalling rviz from `/home/ubuntu/dev/install` and rebuilding?

NoodleOfDeath, 2022-03-09:
> I wonder if this is related to #601 
> 
> Could you try uninstalling rviz from `/home/ubuntu/dev/install` and rebuilding?

I will give it shot and let you know

closed 2022-05-04 by NoodleOfDeath