ros2/rviz #834: Empty grey area instead of 3D grid with rviz2 on WSL

trishmapow, 2022-02-28:
I'm running ROS 2 foxy on WSL Ubuntu 20.04 with the precompiled packages. When I run `rviz2` I get an empty grey area where the 3D grid should be. If I publish e.g. pointcloud data nothing is displayed. I have tried using `export LIBGL_ALWAYS_INDIRECT=0` and also launching rviz2 through an Xrdp session, with no success. Rviz2 does work fine on my dedicated Linux laptop, and I notice also in the original demo video of WSLg here: https://blogs.windows.com/windows-insider/2021/04/21/announcing-windows-10-insider-preview-build-21364/.
[image]

trishmapow, 2022-03-01:
Full log with `-l` below:
```
[INFO] [1646095262.732274097] [rviz2]: Creating resource group General
[INFO] [1646095262.732683793] [rviz2]: Creating resource group OgreInternal
[INFO] [1646095262.732728981] [rviz2]: Creating resource group OgreAutodetect
[INFO] [1646095262.732867311] [rviz2]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[INFO] [1646095262.733170551] [rviz2]: Registering ResourceManager for type Material
[INFO] [1646095262.733305904] [rviz2]: Registering ResourceManager for type Mesh
[INFO] [1646095262.733353965] [rviz2]: Registering ResourceManager for type Skeleton
[INFO] [1646095262.733437457] [rviz2]: MovableObjectFactory for type 'ParticleSystem' registered.
[INFO] [1646095262.733556504] [rviz2]: ArchiveFactory for archive type FileSystem registered.
[INFO] [1646095262.733589227] [rviz2]: DDS codec registering
[INFO] [1646095262.733677757] [rviz2]: ETC codec registering
[INFO] [1646095262.733710669] [rviz2]: Registering ResourceManager for type HighLevelGpuProgram
[INFO] [1646095262.733983765] [rviz2]: Registering ResourceManager for type Compositor
[INFO] [1646095262.734678530] [rviz2]: MovableObjectFactory for type 'Entity' registered.
[INFO] [1646095262.734729392] [rviz2]: MovableObjectFactory for type 'Light' registered.
[INFO] [1646095262.734763970] [rviz2]: MovableObjectFactory for type 'BillboardSet' registered.
[INFO] [1646095262.734806089] [rviz2]: MovableObjectFactory for type 'ManualObject' registered.
[INFO] [1646095262.734853326] [rviz2]: MovableObjectFactory for type 'BillboardChain' registered.
[INFO] [1646095262.734886423] [rviz2]: MovableObjectFactory for type 'RibbonTrail' registered.
[INFO] [1646095262.735690068] [rviz2]: *-*-* OGRE Initialising
[INFO] [1646095262.735751491] [rviz2]: *-*-* Version 1.12.1 (Rhagorthua)
[INFO] [1646095262.735896549] [rviz2]: OverlayElementFactory for type Panel registered.
[INFO] [1646095262.735944107] [rviz2]: OverlayElementFactory for type BorderPanel registered.
[INFO] [1646095262.735976138] [rviz2]: OverlayElementFactory for type TextArea registered.
[INFO] [1646095262.736019686] [rviz2]: Registering ResourceManager for type Font
[INFO] [1646095262.736372277] [rviz2]: Loading library /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
[INFO] [1646095262.748955824] [rviz2]: Installing plugin: GL RenderSystem
[INFO] [1646095262.749119130] [rviz2]: OpenGL Rendering Subsystem created.
[INFO] [1646095262.751378980] [rviz2]: Plugin successfully installed
[INFO] [1646095262.751441023] [rviz2]: Loading library /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/Codec_STBI.so.1.12.1
[INFO] [1646095262.753082286] [rviz2]: stb_image - v2.19 - public domain JPEG/PNG reader
[INFO] [1646095262.753223716] [rviz2]: Supported formats: jpeg,jpg,png,bmp,psd,tga,gif,pic,ppm,pgm,hdr
[INFO] [1646095262.753378726] [rviz2]: CPU Identifier & Features
[INFO] [1646095262.753413193] [rviz2]: -------------------------
[INFO] [1646095262.753420051] [rviz2]:  *   CPU ID: GenuineIntel: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
[INFO] [1646095262.753456779] [rviz2]:  *          SSE: yes
[INFO] [1646095262.753489513] [rviz2]:  *         SSE2: yes
[INFO] [1646095262.753523735] [rviz2]:  *         SSE3: yes
[INFO] [1646095262.753550843] [rviz2]:  *        SSE41: yes
[INFO] [1646095262.753559227] [rviz2]:  *        SSE42: yes
[INFO] [1646095262.753582848] [rviz2]:  *          MMX: yes
[INFO] [1646095262.753616623] [rviz2]:  *       MMXEXT: yes
[INFO] [1646095262.753643424] [rviz2]:  *        3DNOW: no
[INFO] [1646095262.753672491] [rviz2]:  *     3DNOWEXT: no
[INFO] [1646095262.753680895] [rviz2]:  *         CMOV: yes
[INFO] [1646095262.753708048] [rviz2]:  *          TSC: yes
[INFO] [1646095262.753744808] [rviz2]:  *INVARIANT TSC: yes
[INFO] [1646095262.753773320] [rviz2]:  *          FPU: yes
[INFO] [1646095262.753801518] [rviz2]:  *          PRO: yes
[INFO] [1646095262.753810353] [rviz2]:  *           HT: no
[INFO] [1646095262.753835333] [rviz2]: -------------------------
[INFO] [1646095262.753865547] [rviz2]: ******************************
*** Starting GLX Subsystem ***
******************************
[INFO] [1646095262.753991298] [rviz2]: GLX_VERSION = 1.4
[INFO] [1646095262.754044805] [rviz2]: GLX_EXTENSIONS = GLX_ARB_context_flush_control GLX_ARB_create_context GLX_ARB_create_context_no_error GLX_ARB_create_context_profile GLX_ARB_create_context_robustness GLX_ARB_fbconfig_float GLX_ARB_framebuffer_sRGB GLX_ARB_get_proc_address GLX_ARB_multisample GLX_EXT_buffer_age GLX_EXT_create_context_es2_profile GLX_EXT_create_context_es_profile GLX_EXT_fbconfig_packed_float GLX_EXT_framebuffer_sRGB GLX_EXT_import_context GLX_EXT_no_config_context GLX_EXT_swap_control GLX_EXT_swap_control_tear GLX_EXT_texture_from_pixmap GLX_EXT_visual_info GLX_EXT_visual_rating GLX_ATI_pixel_format_float GLX_INTEL_swap_event GLX_MESA_copy_sub_buffer GLX_MESA_multithread_makecurrent GLX_MESA_query_renderer GLX_MESA_swap_control GLX_NV_float_buffer GLX_OML_swap_method GLX_OML_sync_control GLX_SGIS_multisample GLX_SGIX_fbconfig GLX_SGIX_pbuffer GLX_SGIX_visual_select_group GLX_SGI_make_current_read GLX_SGI_swap_control GLX_SGI_video_sync
[INFO] [1646095262.754404766] [rviz2]: Registering ResourceManager for type Texture
[INFO] [1646095262.754598516] [rviz2]: GLRenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 border=none contentScalingFactor=1.000000 currentGLContext=false externalWindowHandle=10485761 left=0 parentWin [... cut ...]
[INFO] [1646095263.920281641] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1646095263.988700829] [rviz2]: Mesh: Loading rviz_sphere.mesh.
[INFO] [1646095264.030007616] [rviz2]: Texture 'SelectionRect0Texture': Loading 1 faces(PF_R8G8B8A8,1x1x1) Internal format is PF_R8G8B8A8,1x1x1.
[INFO] [1646095264.040929436] [rviz2]: Mesh: Loading rviz_cylinder.mesh.
[INFO] [1646095264.045957554] [rviz2]: Mesh: Loading rviz_cone.mesh.
[INFO] [1646095264.076858559] [rviz2]: Using FSAA from GL_ARB_multisample extension.
```

gatboccia, 2022-04-08:
Seeing the same issue in docker Ubuntu 20.04, except on a Linux host. Launching rviz2 around 10+ times does make the grid finally render though, I have not found any significant data in logs, straces, etc to provide any clues.

trishmapow, 2022-04-11:
Trying @gatboccia's suggestion does display various parts of the grid sometimes, but there is a weird persistent effect where the lines and points draw over each other.
[image]

faev999, 2023-02-08:
Hi all,
I experienced the same situation with a similar configuration: WSL2 (ubuntu22) + Docker + ROS2 Humble + RVIZ2. I was using the `ros:humble-ros-core` image and  I managed to fix the problem by switching to the `ros:humble-ros-base` . I tried to get back to the core image and manually add to the docker file the additional dependencies that the base tag supposedly uses, however, the problem came back, and didn't manage to fix it even when adding many other dependencies into the docker file.

faev999, 2023-03-07:
> 
Hi, were you using the core image for Humble?

faev999, 2023-03-07:
> I managed to fix this issue, I am working in a docker container with Ros2 Humble on Ubuntu 22.04. The issue is with openGL and adding `--volume="/dev/dri:/dev/dri" \` to your bash script

Hi, were you using the core image for Humble?

SheaFixstars, 2023-04-14:
Hello, I believe I hit a similar issue. I have tracked it down as far as I can tell to a Ubuntu Mesa regression with Orge and WSL rendering. https://bugs.launchpad.net/ubuntu/+source/mesa/+bug/2004649

Although this is a recent regression not sure if it affects the original report but affects ROS2 Humble release for sure atm.

To get around this I added to my docker image that is based off of ubuntu:22.04 the following to update the mesa libs
```
#Fix broken Ubuntu 22.04 Mesa for WSL2 Docker https://bugs.launchpad.net/ubuntu/+source/mesa/+bug/2004649
RUN apt-get update
RUN apt -y install software-properties-common
RUN add-apt-repository -y ppa:kisak/kisak-mesa
RUN apt-get update
RUN apt-get -y upgrade
```

kaleb236, 2025-02-01:
> Hello, I believe I hit a similar issue. I have tracked it down as far as I can tell to a Ubuntu Mesa regression with Orge and WSL rendering. [...]

this crashed my system I cannot even open any gui anymore. is there any way to uninstall it?