ros2/rviz #836: Compile error cancel marked override but does not override

zflat, 2022-03-02:
It looks like the changes in https://github.com/ros2/rviz/pull/809 have caused errors building. See https://build.ros2.org/job/Rdev__rviz__ubuntu_jammy_amd64/1/consoleText

```
--- stderr: rviz_common
In file included from /tmp/ws/src/rviz/rviz_common/src/rviz_common/transformation/identity_frame_transformer.cpp:32:
/tmp/ws/src/rviz/rviz_common/src/rviz_common/transformation/identity_frame_transformer.hpp:120:3: error: ‘void rviz_common::transformation::IdentityFrameTransformer::cancel(const TransformStampedFuture&)’ marked ‘override’, but does not override
  120 |   cancel(const tf2_ros::TransformStampedFuture & ts_future) override;
      |   ^~~~~~
gmake[2]: *** [CMakeFiles/rviz_common.dir/build.make:2144: CMakeFiles/rviz_common.dir/src/rviz_common/transformation/identity_frame_transformer.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/rviz_common.dir/all] Error 2
```

I am seeing the same compile error with `g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0` on ROS Rolling.

aprotyas, 2022-03-02:
Are you building everything from source? The virtual `cancel` function was added to `tf2_ros::AsyncBufferInterface` in https://github.com/ros2/geometry2/pull/490/ and is available in the latest release (0.22.0).

clalancette, 2022-03-02:
Yes, you need to rebuild both geometry2 and rviz for this to compile now.  We are about to rebuild everything into binary packages; once that is done, this will be fixed.  I'm going to close this one out.

closed 2022-03-02 by clalancette

fzoric8, 2022-06-02:
Hi guys, I'm building `rviz` and `geometry` from source. 

Both repos are on `galactic` branch. 

**Problem still persists.**

clalancette, 2022-06-02:
If both of those repositories are on the galactic branch, it can't possibly be this problem; that API was new in Humble.  If you are still having other problems, please open up a new issue and give us lots of details about how you installed Galactic, the commands you ran to checkout and build, and the errors you are getting.

fzoric8, 2022-06-02:
I mean if I do following: 

```
echo $ROS_DISTRO
```

Output is: 
`galactic`

And I source ROS as `source /opt/ros/galactic/setup.bash`
Which means I have galactic. 

Than If I create new workspace for `rviz` as follows: 
```
mkdir -p rviz_ws/src
cd rviz_ws/src
git clone -b galactic --single-branch https://github.com/ros2/rviz.git
git clone -b galactic --single-branch https://github.com/ros2/geometry2.git 
colcon build --merge-install
```

Build of `rviz` fails with `stderr` output as: 
```
[Processing: rviz_common]                                    
--- stderr: rviz_common                                          
In file included from /home/developer/rviz2_ws/src/rviz/rviz_common/src/rviz_common/transformation/identity_frame_transformer.cpp:32:
/home/developer/rviz2_ws/src/rviz/rviz_common/src/rviz_common/transformation/identity_frame_transformer.hpp:120:3: error: ‘void rviz_common::transformation::IdentityFrameTransformer::cancel(const TransformStampedFuture&)’ marked ‘override’, but does not override
  120 |   cancel(const tf2_ros::TransformStampedFuture & ts_future) override;
      |   ^~~~~~
make[2]: *** [CMakeFiles/rviz_common.dir/build.make:2001: CMakeFiles/rviz_common.dir/src/rviz_common/transformation/identity_frame_transformer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:257: CMakeFiles/rviz_common.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_common [1min 6s, exited with code 2]
                                        
Summary: 19 packages finished [8min 45s]
  1 package failed: rviz_common
  1 package had stderr output: rviz_common
  3 packages not processed
```

So I'm quite sure **problem persists. **

If you're intersted how ROS is installed, you can check Dockerfile [here](https://github.com/osrf/mbzirc/blob/main/docker/mbzirc_sim/Dockerfile).