ros2/rviz #840: :man_farmer:  `rviz_rendering` test regressions for GLSL 4.5

Blast545, 2022-03-03:
New test regression on the linux ci jobs:
```
projectroot.point_cloud_test_target
projectroot.point_cloud_renderable_test_target
rviz_rendering.PointCloudTestFixture.setRenderMode_regenerates_renderables_with_different_size_when_geometry_support_changes
rviz_rendering.PointCloudTestFixture.addPoints_adds_vertices_with_correct_geometry_when_called
rviz_rendering.PointCloudRenderableTestFixture.renderable_contains_a_correctly_filled_buffer
```
Reference: https://ci.ros2.org/view/nightly/job/nightly_linux_debug/2223/
https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_debug/1895/

The package failing is `rviz_rendering` and the failing tests are: 

https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/test/rviz_rendering/objects/point_cloud_test.cpp#L267
https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/test/rviz_rendering/objects/point_cloud_test.cpp#L298
https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/test/rviz_rendering/objects/point_cloud_renderable_test.cpp#L91

The underlying problem is that the opengl version used is 4.5(GLSL 4.5)  instead of Opengl 3.1 (GLSL 1.4), so probably these failing tests will have to be adjusted or the opengl version has to be fixed. All the tests seem to be covering the cases of GLSL>=120 and GLSL>=150, for the case of GLSL>=4.5 the tests seem to expect a similar behavior from the older version. 

Tagging package maintainers to get feedback. @jacobperron @mjeronimo

jacobperron, 2022-03-03:
Is it just the tests that need to be updated or is there also a change needed related to materials? Here:

https://github.com/ros2/rviz/blob/e4a0d4a9f1c42ea9dd93c971b5e36ef356a4110d/rviz_rendering/src/rviz_rendering/render_system.cpp#L380-L388

I'm not that familiar with the rendering code, but I guess we should confirm that point clouds are still being rendered with the newer opengl version (in addition to fixing tests).

Blast545, 2022-03-04:
@jacobperron I am not that familiar with the rendering code, do you know someone we could tag to take a look?

> or is there also a change needed related to materials? Here:

I didn't see any errors popping out from that part of the code, so I think that only the tests have to be updated.

Blast545, 2022-03-15:
:man_farmer: I took some time digging into this as it's affecting various jobs on the buildfarm. I think there problem is not related to a change in the number of vertices available in different OpenGL versions, I think it's related to the <<rendering modes available>>

i.e. These type of comparisons:

https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/test/rviz_rendering/objects/point_cloud_test.cpp#L261-#L266

Are not a direct relationship to the opengl version but to the rendering mode being used:
https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/src/rviz_rendering/objects/point_cloud.cpp#L141-L158

However we can't access this variable:
https://github.com/ros2/rviz/blob/f5f7c6a548b573959363307e9ad3ffab34acd93d/rviz_rendering/include/rviz_rendering/objects/point_cloud.hpp#L321
As it's private to the point_coud object. Does it make sense to create a getter for this class variable? that way we could change the check for opengl version to check for the rendering mode used.

jacobperron, 2022-03-15:
@Blast545 Thanks for digging into it! Sorry, I haven't had the cycles.

I think adding a getter for that member flag would be fine.

PR #843 (2022-03-17): Make getVerticesPerPoint method public and improve tests
Should close #840 

I didn't have to extend existing API because it seems there's already API available to get the number of vertices being used after the render mode is set.

Opening as draft until I check with CI.
closed 2022-03-17 by clalancette