ros2/rviz #842: Rviz Camera plugin render distance is hardcoded at 100 meters

paulerikf, 2022-03-15:
Noticed the rviz camera plugin's overlay plugin doesn't render anything past 100 meters. Modifying the value of `far_plane` in camera_display.cpp changes the render distance.
https://github.com/ros2/rviz/blob/e4a0d4a9f1c42ea9dd93c971b5e36ef356a4110d/rviz_default_plugins/src/rviz_default_plugins/displays/camera/camera_display.cpp#L596

Is there a reason it's currently hardcoded to 100 m? Could we bump that value up, or make it configurable?

clalancette, 2022-03-22:
I'd say let's make it configurable, keeping the default at 100 meters.  There is probably some cost we have to pay (in processing time, or accuracy, or something) for larger values, so I think it makes sense to leave the default where it is.

jacobperron, 2022-03-28:
Making this configurable makes sense to me. @paulerikf If you'd like to propose a PR, I'm happy to review it.

paulerikf, 2022-03-29:
Alright, should have some time to set up a PR next week. Thanks!

closed 2022-04-07 by jacobperron