ros2/rviz #845: ERROR compiling rviz_ogre_vendor: Could NOT find Patch (missing: Patch_EXECUTABLE)

SpaceView, 2022-03-28:
compile ROS2 galactic in windows 10
however I got the following error for rviz_ogre_vendor
content in stdout_stderr.log is given as follows,

-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- The C compiler identification is MSVC 19.29.30139.0
-- The CXX compiler identification is MSVC 19.29.30139.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 1.1.5 (C:/dev/ros2_galactic/install/share/ament_cmake/cmake)
-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Patch (missing: Patch_EXECUTABLE)
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPatch.cmake:70 (find_package_handle_standard_args)
  CMakeLists.txt:157 (find_package)
  CMakeLists.txt:214 (build_ogre)

-- Configuring incomplete, errors occurred!
See also "C:/dev/ros2_galactic/build/rviz_ogre_vendor/CMakeFiles/CMakeOutput.log".

jacobperron, 2022-03-28:
Is Git installed on your system? CMake is not able to find the patch executable, so it could also be that it is installed in an unexpected location.

I would suggest asking this kind of question on ROS answers, but I found the someone running into the same problem without an answer: https://answers.ros.org/question/383548/could-not-find-patch-missing-patch_executable-rviz_ogre_vendor/

clalancette, 2022-04-14:
No response in over two weeks, so I'm going to close this out.  If you are still running into problems, please feel free to supply the requested information and reopen.

closed 2022-04-14 by clalancette